SpinalHDL/SpinalHDL #1276: Port names get lost on dev branch

KireinaHoro, 2024-01-11:
I have a simple blackbox that works in `1.9.0`:

```scala
import spinal.core._
import spinal.lib._
import spinal.lib.bus.amba4.axis._
import spinal.lib.bus.amba4.axis.Axi4Stream.Axi4Stream

case class AxiStreamFrameLen(axisConfig: Axi4StreamConfig, lenWidth: Int = 16) extends BlackBox {
  assert(clockDomain.config.resetKind == SYNC, "verilog-axis requires synchronous reset")

  val keepWidth = axisConfig.dataWidth // BYTES
  val generic = new Generic {
    val DATA_WIDTH = axisConfig.dataWidth * 8
    val KEEP_ENABLE = axisConfig.useKeep
    val KEEP_WIDTH = keepWidth
    val LEN_WIDTH = lenWidth
  }
  val modName = "axis_frame_len"
  setBlackBoxName(modName)

  val io = new Bundle {
    val clk = in Bool()
    val rst = in Bool()

    val monitor_axis = new Bundle {
      val tkeep = in UInt (keepWidth bits)
      val tvalid = in Bool()
      val tready = in Bool()
      val tlast = in Bool()
    }

    val frame_len = master(Flow(UInt(lenWidth bits)))
  }

  def driveFrom(axis: Axi4Stream): Unit = {
    io.monitor_axis.tkeep := axis.payload.keep.asUInt
    io.monitor_axis.tvalid := axis.valid
    io.monitor_axis.tready := axis.ready
    io.monitor_axis.tlast := axis.payload.last
  }

  mapCurrentClockDomain(io.clk, io.rst)
  noIoPrefix()
  addPrePopTask { () =>
    Component.current.getAllIo.find(_.getName == "frame_len_payload").get.setName("frame_len")
  }

  addRTLPath(axisRTLFile(modName))
}

object FrameLenVerilog extends App {
  Config.spinal.generateVerilog(new Component {
    val inst = AxiStreamFrameLen(Axi4StreamConfig(128))
    val io = new Bundle {
      val out = master(inst.io.frame_len.clone)
    }
    io.out := inst.io.frame_len
    inst.io.monitor_axis.assignDontCare()
  })
}
```

It produces the following Verilog:

```verilog
// Generator : SpinalHDL v1.9.0    git head : 7d30dbacbd3aa1be42fb2a3d4da5675703aae2ae
// Component : unamed
// Git hash  : bd70e3adfa36b1ed6e30202ea319eacd4befa844

`timescale 1ns/1ps

module unamed (
  input               reset,
  input               clk,
  output              io_out_valid,
  output     [15:0]   io_out_payload
);

  wire                inst_frame_len_valid;
  wire       [15:0]   inst_frame_len;

  axis_frame_len #(
    .DATA_WIDTH(1024),
    .KEEP_ENABLE(1'b0),
    .KEEP_WIDTH(128),
    .LEN_WIDTH(16)
  ) inst (
    .clk                 (clk                                                                                                                                  ), //i
    .rst                 (reset                                                                                                                                ), //i
    .monitor_axis_tkeep  (128'bxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx), //i
    .monitor_axis_tvalid (1'bx                                                                                                                                 ), //i
    .monitor_axis_tready (1'bx                                                                                                                                 ), //i
    .monitor_axis_tlast  (1'bx                                                                                                                                 ), //i
    .frame_len_valid     (inst_frame_len_valid                                                                                                                 ), //o
    .frame_len           (inst_frame_len[15:0]                                                                                                                 )  //o
  );
  assign io_out_valid = inst_frame_len_valid;
  assign io_out_payload = inst_frame_len;

endmodule
```

But with the latest `dev` branch (058f199e1b5e50321e4b35f04abc5abc5c255b6a), `getAllIo` returns signals with `???` as the name:

```
(toplevel/[AxiStreamFrameLen]/??? : in Bool)                                                                                                                                                 
(toplevel/[AxiStreamFrameLen]/??? : in Bool)                                                                                                                                                 
(toplevel/[AxiStreamFrameLen]/??? : in UInt[128 bits])                                                                                                                                       
(toplevel/[AxiStreamFrameLen]/??? : in Bool)                                                                                                                                                 
(toplevel/[AxiStreamFrameLen]/??? : in Bool)                                                                                                                                                 
(toplevel/[AxiStreamFrameLen]/??? : in Bool)                                                                                                                                                 
(toplevel/[AxiStreamFrameLen]/??? : out Bool)                                                                                                                                                
(toplevel/[AxiStreamFrameLen]/??? : out UInt[16 bits])                                                                                                                                       
(toplevel/[AxiStreamFrameLen]/??? : out Bool)                                                                                                                                                
(toplevel/[AxiStreamFrameLen]/??? : out UInt[16 bits])  
```

Is there a recent change that led to this regression?

KireinaHoro, 2024-01-11:
Ah never mind, I didn't add the IDSL compiler plugins correctly.

closed 2024-01-11 by KireinaHoro

KireinaHoro, 2024-01-12:
@Dolu1990 in hindsight, would it be a good idea to have the elaboration fail when the plugin is not loaded?  Could save a lot of time in case someone forgot to add the idsl plugin ivy dependency...

Dolu1990, 2024-01-12:
Yes it would be a good idea, thing is, i don't know how to do it XD
Maybe by having the plugin overriding a dummy function in the Component class, allowing runtime detection, or something like that.