SpinalHDL/SpinalHDL #1278: Question about sim data sample and drive delay

developfpga, 2024-01-12:
Hi,

I got puzzled about sim data sample and drive delay. Let's check the wave below
[image]

Checkint io_tx_dReady and io_tx_eop. io_tx_dReady is driving by spinal sim, io_tx_eop is driving by spinal rtl register.

First question, io_tx_eop has some delay than io_txClock60M, but io_tx_dReady has not. So where is the delay comes from? Or how can I add some delay to RTL register?
Second question, in normal verilog sense, io_tx_dReady asserted at the first posedge of clk, and io_tx_dReady can be checked to be asserted at the next posedge of clk, but in spinal sim, io_tx_dReady is checked to be asserted the same posedge of clk.

developfpga, 2024-01-12:
the spinal sim code
[image]

Dolu1990, 2024-01-12:
Hi,

Normaly there shouldn't have been a delay.
Can you provide a self contained way for us to reproduce the case ?

developfpga, 2024-01-17:
Sorry for delayed response, here is the test code.
The delay of io_tx_sop seems to be related to two settings.
One is clock60M, which is divided from clk; the other one the VCS simulator. It's good with Iverilator

import org.scalatest.funsuite.AnyFunSuite
import spinal.core._
import spinal.lib._
import spinal.core.sim._
import spinal.sim.VCSFlags

import scala.collection.mutable
import scala.math.BigInt
import scala.util.Random

case class Dut() extends Component {
	val io = new Bundle {
		val tx = slave(SendPacket())
		val clock60M = out Bool()
		val reset60M = out Bool()
	}
	val count = Reg(UInt(2 bits)) init(0)
	count := count + 1
	val clock60M = Reg(Bool()) init(False)
	val reset60M = Reg(Bool()) init(True)
	when (count(0, 2 bits).andR) {
		clock60M := ~clock60M
	}
	when (clock60M) {
		reset60M.clear()
	}
	val clock60MDomain = ClockDomain(io.clock60M, io.reset60M)
	val clock60MArea = new ClockingArea(clock60MDomain) {
		val dReady = Reg(Bool()) init(False)
		when(io.tx.sop) {
			dReady.set()
		} elsewhen (io.tx.eop) {
			dReady.clear()
		}
	}

	io.tx.dReady := clock60MArea.dReady
	io.tx.dEmpty := ~clock60MArea.dReady
	io.clock60M := clock60M
	io.reset60M := reset60M

}
class Tester() extends AnyFunSuite {
	var compiled: SimCompiled[Dut] = null

	object VcsSimConfig {
		val flags = VCSFlags(
			compileFlags = List("-kdb"),
			elaborateFlags = List("-kdb")
		)

		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
			.withVCS(flags)
			.withFSDBWave
			.withVCSSimSetup(
				setupFile = "src/uvm/synopsys_sim.setup",
				beforeAnalysis = () => {
					println("Hello VCS")
				}
			)
	}
	object IverilatorSimConfig {
		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim").addIncludeDir("./rtl/include/")
	}
	test("compiled") {
		compiled = VcsSimConfig().compile {
			val dut = new Dut()
			dut
		}
	}

	test("bringup") {
		compiled.doSim("bringup") { dut =>

			val sendPacketDriver = SendPacketDriver(dut.clock60MDomain, dut.io.tx)

			dut.clockDomain.forkStimulus(2082 ps)
			sendPacketDriver.clearPort()
			dut.clockDomain.waitSampling()
			val sendPacket = new Array[Byte](10)
			Random.nextBytes(sendPacket)
			sendPacketDriver.push(sendPacket)
			0.until(100).map(_ => dut.clock60MDomain.waitSampling())
		}
	}
}

// d is two cycle later than dReady
// dEmpty means all data in tx buffer is sent
case class SendPacket() extends Bundle with IMasterSlave {

	val d = Bits (8 bits)
	val sop = Bool()
	val eop = Bool()
	val dReady = Bool()
	val dEmpty = Bool()

	override def asMaster(): Unit = {
		out(d, sop, eop)
		in(dReady, dEmpty)
	}
}
case class SendPacketDriver (clockDomain: ClockDomain, tx: SendPacket) {
	var packetQueue = mutable.Queue[Array[Byte]]()
	var gapBetweenPackets = 0

	def clear() {
		this.packetQueue.clear()
	}

	def push(packet: Array[Byte], aux: Int = 0) {
		this.packetQueue.enqueue(packet)
	}

	def getNoOfPackets(): Int = {
		this.packetQueue.length
	}

	def driver(once: Boolean = false): Unit = {
		var sendPacket: Array[Byte] = Array()
		var sendData: BigInt = 0
		var break: Boolean = false
		var dReadyR1, dReadyR2 = false
		var sendByteCount = 0

		tx.d #= 0
		tx.sop #= false
		tx.eop #= false

		clockDomain.waitRisingEdge()
		while(!break) {
			if(this.getNoOfPackets() != 0){
				sendPacket = this.packetQueue.dequeue()
				tx.sop #= true
				tx.eop #= false
				clockDomain.waitRisingEdge()
				tx.sop #= false
				clockDomain.waitRisingEdge()
				dReadyR1 = false
				dReadyR2 = false
				sendByteCount = 0
				if (sendPacket.size != 0) {
					while (sendPacket.size > sendByteCount) {
						dReadyR2 = dReadyR1
						dReadyR1 = tx.dReady.toBoolean
						//						if (dReadyR2) {
						if (tx.dReady.toBoolean) {
							tx.d #= BigInt(sendPacket(sendByteCount) & 0xff)
							sendByteCount = sendByteCount + 1
						}
						clockDomain.waitRisingEdge()
					}
				} else {
					tx.d #= 0
					tx.sop #= false
					tx.eop #= false
					sendData = 0
					clockDomain.waitRisingEdge()
					while (!tx.dReady.toBoolean) {
						clockDomain.waitRisingEdge()
					}
				}
				tx.eop #= true
				clockDomain.waitRisingEdge()
				while (!tx.dEmpty.toBoolean) {
					clockDomain.waitRisingEdge()
				}
				tx.eop #= false
				if (once) {
					break = true
				}
			} else {
				tx.d #= 0
				tx.sop #= false
				tx.eop #= false
				clockDomain.waitRisingEdge()
			}
		}
	}

	var state = 0
	var sendPacket: Array[Byte] = Array()
	var dReadyR1, dReadyR2 = false
	var sendByteCount = 0
	tx.d #= 0
	tx.sop #= false
	tx.eop #= false

	def driverOnSampling(): Unit = {
		state match {
			case 0 => {
				tx.sop #= false
				tx.eop #= false
				dReadyR1 = false
				dReadyR2 = false
				sendByteCount = 0
				if(this.getNoOfPackets() != 0) {
					sendPacket = this.packetQueue.dequeue()
					if (sendPacket.length != 0) {
						tx.sop #= true
						tx.eop #= false
						state = 1
					}
				}
			}
			case 1 => {
				tx.sop #= false
				state = 2
			}
			case 2 => {
				if (dReadyR2) {
					//				if (tx.dReady.toBoolean) {
					tx.d #= BigInt(sendPacket(sendByteCount) & 0xff)
					sendByteCount = sendByteCount + 1
					if (sendPacket.length == sendByteCount) {
						state = 3
					}
				}
				dReadyR2 = dReadyR1
				dReadyR1 = tx.dReady.toBoolean
			}
			case 3 => {
				tx.eop #= true
				if (tx.dEmpty.toBoolean) {
					tx.eop #= false
					state = 0
				}
			}
		}
	}
	clockDomain.onSamplings(driverOnSampling())

	def clearPort(): Unit = {
		tx.d #= 0
		tx.sop #= false
		tx.eop #= false
	}
}

developfpga, 2024-01-17:
What about my second question?

> Second question, in normal verilog sense, io_tx_dReady asserted at the first posedge of clk, and io_tx_dReady can be checked to be asserted at the next posedge of clk, but in spinal sim, io_tx_dReady is checked to be asserted the same posedge of clk.

Actually, I want io_tx_eop asserted one more cycle.
The wave:
VCS:
[image]
Iverilator:
[image]

Dolu1990, 2024-01-17:
Hi

You code sample is mached up by github.

Can you send it between : 

\`\`\`scala
your scala code
\`\`\`

?

Also, that's a lot of code, i'm not sure to understand what surprise you, can you provide a simple sample of code which reproduce the behaviour you don't understand ? 

One trick to help you, maybe to add some println(s"XXX ${simTime()}) at some place in your code to localise things

developfpga, 2024-01-18:
Hi Dolu,

I tried two ways of driving signals at SendPacketDriver, one is clockDomain.waitRisingEdge() at method driver, the other one is clockDomain.onSamplings at method driverOnSampling. Now the test use driverOnSampling default.

Please check "println(s"get dEmpty ${simTime()}")"

> [Progress] Start Dut bringup simulation with seed 972763248
get dEmpty 307095
Unexpected termination of the simulation
           V C S   S i m u l a t i o n   R e p o r t 

[image]

```scala

import org.scalatest.funsuite.AnyFunSuite
import spinal.core._
import spinal.lib._
import spinal.core.sim._
import spinal.sim.VCSFlags

import scala.collection.mutable
import scala.math.BigInt
import scala.util.Random

case class Dut() extends Component {
	val io = new Bundle {
		val tx = slave(SendPacket())
		val clock60M = out Bool()
		val reset60M = out Bool()
	}
	val count = Reg(UInt(2 bits)) init(0)
	count := count + 1
	val clock60M = Reg(Bool()) init(False)
	val reset60M = Reg(Bool()) init(True)
	when (count(0, 2 bits).andR) {
		clock60M := ~clock60M
	}
	when (clock60M) {
		reset60M.clear()
	}
	val clock60MDomain = ClockDomain(io.clock60M, io.reset60M)
	val clock60MArea = new ClockingArea(clock60MDomain) {
		val dReady = Reg(Bool()) init(False)
		when(io.tx.sop) {
			dReady.set()
		} elsewhen (io.tx.eop) {
			dReady.clear()
		}
	}

	io.tx.dReady := clock60MArea.dReady
	io.tx.dEmpty := ~clock60MArea.dReady
	io.clock60M := clock60M
	io.reset60M := reset60M

}
class Tester() extends AnyFunSuite {
	var compiled: SimCompiled[Dut] = null

	object VcsSimConfig {
		val flags = VCSFlags(
			compileFlags = List("-kdb"),
			elaborateFlags = List("-kdb")
		)

		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
			.withVCS(flags)
			.withFSDBWave
[... 163 lines cut ...]
					}
				}
				dReadyR2 = dReadyR1
				dReadyR1 = tx.dReady.toBoolean
			}
			case 3 => {
				tx.eop #= true
				if (tx.dEmpty.toBoolean) {
					println(s"get dEmpty ${simTime()}")
					tx.eop #= false
					state = 0
				}
			}
		}
	}
	clockDomain.onSamplings(driverOnSampling())

	def clearPort(): Unit = {
		tx.d #= 0
		tx.sop #= false
		tx.eop #= false
	}
}

```

developfpga, 2024-01-19:
yes, I tried, It works well with verilator.

developfpga, 2024-01-19:
I simplified the code, please try it

```scala

import org.scalatest.funsuite.AnyFunSuite
import spinal.core._
import spinal.lib._
import spinal.core.sim._
import spinal.sim.VCSFlags

import scala.collection.mutable
import scala.math.BigInt
import scala.util.Random

case class Dut() extends Component {
	val io = new Bundle {
		val tx = slave(SendPacket())
		val clock60M = out Bool()
		val reset60M = out Bool()
	}
	val count = Reg(UInt(2 bits)) init(0)
	count := count + 1
	val clock60M = Reg(Bool()) init(False)
	val reset60M = Reg(Bool()) init(True)
	when (count(0, 2 bits).andR) {
		clock60M := ~clock60M
	}
	when (clock60M) {
		reset60M.clear()
	}
	val clock60MDomain = ClockDomain(io.clock60M, io.reset60M)
	val clock60MArea = new ClockingArea(clock60MDomain) {
		val dReady = Reg(Bool()) init(False)
		when(io.tx.sop) {
			dReady.set()
		} elsewhen (io.tx.eop) {
			dReady.clear()
		}
	}

	io.tx.dReady := clock60MArea.dReady
	io.tx.dEmpty := ~clock60MArea.dReady
	io.clock60M := clock60M
	io.reset60M := reset60M

}
class Tester() extends AnyFunSuite {
	var compiled: SimCompiled[Dut] = null

	object VcsSimConfig {
		val flags = VCSFlags(
			compileFlags = List("-kdb"),
			elaborateFlags = List("-kdb")
		)

		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
			.withVCS(flags)
			.withFSDBWave
			.withVCSSimSetup(
				setupFile = "src/uvm/sv/synopsys_sim.setup",
				beforeAnalysis = () => {
					println("Hello VCS")
				}
			)
	}
	test("compiled") {
		compiled = VcsSimConfig().compile {
			val dut = new Dut()
			dut
		}
	}

	test("bringup") {
		compiled.doSim("bringup") { dut =>

			val sendPacketDriver = SendPacketDriver(dut.clock60MDomain, dut.io.tx)

			dut.clockDomain.forkStimulus(2082 ps)
			sendPacketDriver.clearPort()
			dut.clockDomain.waitSampling()
			val sendPacket = new Array[Byte](10)
			Random.nextBytes(sendPacket)
			sendPacketDriver.push(sendPacket)
			0.until(100).map(_ => dut.clock60MDomain.waitSampling())
		}
	}
}

// d is two cycle later than dReady
// dEmpty means all data in tx buffer is sent
case class SendPacket() extends Bundle with IMasterSlave {

	val d = Bits (8 bits)
	val sop = Bool()
	val eop = Bool()
	val dReady = Bool()
	val dEmpty = Bool()

	override def asMaster(): Unit = {
		out(d, sop, eop)
		in(dReady, dEmpty)
	}
}
case class SendPacketDriver (clockDomain: ClockDomain, tx: SendPacket) {
	var packetQueue = mutable.Queue[Array[Byte]]()
	var gapBetweenPackets = 0

	def clear() {
		this.packetQueue.clear()
	}

	def push(packet: Array[Byte], aux: Int = 0) {
		this.packetQueue.enqueue(packet)
	}

	def getNoOfPackets(): Int = {
		this.packetQueue.length
	}

	var state = 0
	var sendPacket: Array[Byte] = Array()
	var dReadyR1, dReadyR2 = false
	var sendByteCount = 0
	tx.d #= 0
	tx.sop #= false
	tx.eop #= false

	def driverOnSampling(): Unit = {
		state match {
			case 0 => {
				tx.sop #= false
				tx.eop #= false
				dReadyR1 = false
				dReadyR2 = false
				sendByteCount = 0
				if(this.getNoOfPackets() != 0) {
					sendPacket = this.packetQueue.dequeue()
					if (sendPacket.length != 0) {
						tx.sop #= true
						tx.eop #= false
						state = 1
					}
				}
			}
			case 1 => {
				tx.sop #= false
				state = 2
			}
			case 2 => {
				if (dReadyR2) {
					//				if (tx.dReady.toBoolean) {
					tx.d #= BigInt(sendPacket(sendByteCount) & 0xff)
					sendByteCount = sendByteCount + 1
					if (sendPacket.length == sendByteCount) {
						state = 3
					}
				}
				dReadyR2 = dReadyR1
				dReadyR1 = tx.dReady.toBoolean
			}
			case 3 => {
				tx.eop #= true
				if (tx.dEmpty.toBoolean) {
					println(s"get dEmpty ${simTime()}")
					tx.eop #= false
					state = 0
				}
			}
		}
	}
	clockDomain.onSamplings(driverOnSampling())

	def clearPort(): Unit = {
		tx.d #= 0
		tx.sop #= false
		tx.eop #= false
	}
}
```

developfpga, 2024-01-19:
I write another system verilog testbench to test the Dut, which is generated by class Dut.
The same logic as driverOnSampling of spinal test.
The wave:
[image]
There is no unexpected delay.
And more importantly, the signal tx_eop asserted for 2 cycles, this is what I expected. Not like the test in spinal sim, which only asserted for 1 cycle

``` systemverilog
`timescale 1ns/1ps
module tb_dut();

	logic [7:0] tx_d		;
	logic tx_sop	;
	logic tx_eop	;
	logic tx_dReady	;
	logic tx_dEmpty	;
	logic clock60M	;
	logic reset60M	;
	logic clk		;
	logic reset		;

	Dut u_Dut (
		.io_tx_d		(tx_d		),
		.io_tx_sop		(tx_sop		),
		.io_tx_eop		(tx_eop		),
		.io_tx_dReady	(tx_dReady	),
		.io_tx_dEmpty	(tx_dEmpty	),
		.io_clock60M	(clock60M	),
		.io_reset60M	(reset60M	),
		.clk			(clk		),
		.reset			(reset		)
	);

	initial begin
		clk = 1'b0;
		reset = 1'b1;

		#100;
		reset = 1'b0;
	end
	always #2.082 clk = ~clk;

	logic [1:0] state;
	logic [2:0] tx_count;
	logic dReady_r1;
	always @(posedge clock60M or posedge reset60M) begin
		if (reset60M) begin
			tx_sop <= 1'b0;
			tx_eop <= 1'b0;
			tx_d <= 8'b0;
			state <= 2'b0;
			tx_count <= 3'b0;
			dReady_r1 <= 1'b0;
		end else begin
			tx_count <= 3'b0;
			dReady_r1 <= tx_dReady;
			case (state)
				2'd0: begin
					tx_sop <= 1'b1;
					tx_eop <= 1'b0;
					state <= 2'd1;
				end
				2'd1: begin
					tx_sop <= 1'b0;
					state <= 2'd2;
				end
				2'd2: begin
					if (dReady_r1) begin
						tx_count <= tx_count + 1;
						tx_d <= tx_d + 1;
					end else begin
						tx_count <= tx_count;
					end
					if (dReady_r1 & &tx_count) begin
						state <= 2'd3;
					end
				end
				2'd3: begin
					tx_eop <= 1'b1;
					if (tx_dEmpty) begin
						tx_eop <= 1'b0;
					end
				end
			endcase
		end
	end

endmodule
```
The VCS script:
``` shell
vlogan -full64 +v2k -sverilog -work work -lca -kdb +define+SVT_FSDB_ENABLE tb_dut.sv
vlogan -full64 -work work -lca -kdb +define+SVT_FSDB_ENABLE ./Dut.v

vcs -full64 -R -gui -debug_all tb_dut

```

developfpga, 2024-01-29:
@Dolu1990 I simpified the test

```scala
test("bringup") {
		compiled.doSim("bringup") { dut =>

			dut.io.tx.sop #= false
			dut.io.tx.eop #= false
			dut.io.tx.d #= 0

			dut.clockDomain.forkStimulus(2082 ps)
			dut.clockDomain.waitSampling()

			dut.clock60MDomain.waitSampling()
			dut.clock60MDomain.waitSampling()
			dut.clock60MDomain.waitSampling()

			// first assert sop
			dut.io.tx.sop #= true
			dut.clock60MDomain.waitSampling()
			dut.io.tx.sop #= false
			dut.clock60MDomain.waitSampling(8)

			// second assert eop
			dut.io.tx.eop #= true
			dut.clock60MDomain.waitSampling()

			// then check dReady
			while (dut.io.tx.dReady.toBoolean) {
				dut.clock60MDomain.waitSampling()
			}

			// finally clear eop
			dut.io.tx.eop #= false

			0.until(100).map(_ => dut.clock60MDomain.waitSampling())

		}
	}
```
the wave updated:
[image]

Actually, there are two questions from the beginning. And the second question is more important to me.
One is the signal delay caused by VCS backend sim, the other one is when the sim simulation data?

About the second question, you can check the eop signal of two wave, one is the wave above(which asserted for one cycle), the other one is the system verilog test result(which asserted for two cycle, and this makes more sense from my experience).

Thanks

developfpga, 2024-02-07:
@wswslzp 
I tried setting timePrecision option to be 1ps like this, but it does not work.
```scala

import org.scalatest.funsuite.AnyFunSuite
import spinal.core._
import spinal.lib._
import spinal.core.sim._
import spinal.sim.VCSFlags

import scala.collection.mutable
import scala.math.BigInt
import scala.util.Random

class Tester1() extends AnyFunSuite {
	var compiled: SimCompiled[Dut] = null

	object VcsSimConfig {
		val flags = VCSFlags(
			compileFlags = List("-kdb"),
			elaborateFlags = List("-kdb")
		)

		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
			.withVCS(flags)
			.withFSDBWave
			.withVCSSimSetup(
				setupFile = "src/uvm/sv/synopsys_sim.setup",
				beforeAnalysis = () => {
					println("Hello VCS")
				}
			)
			.withTimePrecision(1 ps)
	}
	test("compiled") {
		compiled = VcsSimConfig().compile {
			val dut = new Dut()
			dut
		}
	}

	test("bringup") {
		compiled.doSim("bringup") { dut =>

			dut.io.tx.sop #= false
			dut.io.tx.eop #= false
			dut.io.tx.d #= 0

			dut.clockDomain.forkStimulus(2082 ps)
			dut.clockDomain.waitSampling()

			dut.clock60MDomain.waitSampling()
			dut.clock60MDomain.waitSampling()
			dut.clock60MDomain.waitSampling()

			// first assert sop
			dut.io.tx.sop #= true
			dut.clock60MDomain.waitSampling()
			dut.io.tx.sop #= false
			dut.clock60MDomain.waitSampling(8)

			// second assert eop
			dut.io.tx.eop #= true
			dut.clock60MDomain.waitSampling()

			// then check dReady
			while (dut.io.tx.dReady.toBoolean) {
				dut.clock60MDomain.waitSampling()
			}

			// finally clear eop
			dut.io.tx.eop #= false

			0.until(100).map(_ => dut.clock60MDomain.waitSampling())
			
		}
	}
}
```

developfpga, 2024-02-07:
and if I set clock period to 10 ns, it still has the same problem.
[image]

Dolu1990, 2024-02-07:
So, to get again some context, in the wave in the above message, we can see that when the testbench write io_tx_eop, the io_tx_eop value on the wave is update is update in a delayed manner. (not syncronised with clock60MDomain.waitSampling)

Seems like the simulator wait the next time event / delta cycle (clk falling edge)

wswslzp, 2024-02-08:
> and if I set clock period to 10 ns, it still has the same problem. [...]

Currently I don't have VCS env so I cannot help debug further. But from the waveform, your signals are seemed to be sync with **the negedge of clk**. 

If you can help provide these two clock domains' configuration, it may help you debug.

developfpga, 2024-02-19:
Actually, the second question is more important to me. You can check the eop signal of two wave, one is the wave above(which asserted for one cycle), the other one is the system verilog test result(which asserted for two cycle, and this makes more sense from my experience).

> The wave: [...]

> the wave updated: [...]

closed 2024-04-02 by developfpga