SpinalHDL/SpinalHDL #1280: Reg(False) / Reg(True) are valid statements

dnltz, 2024-01-12:
I recently tried to create a Register with a default Bool value like
```
val first = RegInit(False)
```
but accidentally used `Reg` intead of `RegInit`. It seems like SpinalHDL interprets True/False as Bool and it passed all internal checks. I only realized my signal's default value sometimes changes in the simulation.

Is this behavior intended?

Can we forbid the following two statements or at least throw an warning?
```
val first = Reg(True)
val last = Reg(False)
```

Dolu1990, 2024-01-12:
Hi,

Yes, basicaly Reg(x) is equivalent to Reg(cloneOf(x))
The intend is that for instance, if you have some signal, and you want to create a register of the same type, you can just do : 
Reg(myReferenceSignal).

Maybe, Reg(directLiteral) should be forbiden for satefy, as it is likely not wanted.

dnltz, 2024-01-13:
Agree, in my opinion True/False create the impression of an default value but I have no if there's a use-case for that.

I can try to forbid the directLiteral value, if you dont mind.

Dolu1990, 2024-01-13:
Sure ^^
I m curious how many times it will trigger the issue in the code base XD