SpinalHDL/SpinalHDL #1282: Blackbox io generate Bits from Vec as Data

oletf, 2024-01-15:
Hi,

I'm having trouble figuring out how I could have a Blackbox's  Vec[Data] generated as its `Vec.asBits` equivalent after elaboration

```scala
case class MyBlackBox(n: Int = 2) extends BlackBox {
  val io = new Bundle {
    val sStream = slave Stream(Fragment(Vec.fill(n)(Bits(32 bits))))
    val mStream = master Stream(Fragment(Vec.fill(n)(Bits(32 bits))))
  }
}
```
gets properly generated as
```vhdl
component MyBlackBox is
    port( 
     -- ...
      sStream _payload_fragment_0 : in std_logic_vector; -- (32 bits)
      sStream _payload_fragment_1 : in std_logic_vector; -- (32 bits)
     -- [...]
      mStream _payload_fragment_0 : out std_logic_vector; -- (32 bits)
      mStream _payload_fragment_1 : out std_logic_vector; -- (32 bits)
    );
  end component;
```
but I'd need to have
```vhdl
component MyBlackBox is
    port( 
     -- ...
      sStream _payload_fragment : in std_logic_vector; -- (64 bits)
     -- [...]
      mStream _payload_fragment : out std_logic_vector; -- (64 bits)
    );
  end component;
```

I found https://github.com/SpinalHDL/SpinalHDL/issues/1258#issuecomment-1849942630 which seems to be exactly what I need,
but since `InOutVecToBits` isn't `Data` it cannot be contained in `Fragment`.

Any hint on which way I could achieve this ?

Thanks forward

andreasWallner, 2024-01-16:
Since every Fragment comes with it's own `last` signal - what would you expect to happen with those? Should both still be available?
In any case the simple workaround is a small SpinalHDL wrapper. I normally do that for components with weird interfaces: make the blackbox exactly as the component is, most of the time w/o Stream/Flow etc. and then make a wrapper around it where I connect the SpinalHDL types accordingly. I feel it's simpler then the rework step.

oletf, 2024-01-16:
I edited the original message to add the `last` signals for this specific case. 
Here, as in the Spinal io port description, there is only one last per stream since I need a `Fragment(Vec)` 
(and not a `Vec(Fragment)` (in which case that would make one `last` per element indeed)).
The component I have to interface with is exactly like a `Stream(Fragment(Bits(64 bits)))` with `ready`/`valid` and `last`, it's just the Fragment which is actually 2 `SInt(32 bits)` packed together, so it'd be easier to use from Spinal.

The wrapper solution seems very interesting and might be what I actually need, 
definitely feels simpler than the rework step to me as well.

I don't really know how to implement this, would you happen to have a simple example of how you'd achieve this ?
(even an example with just a single `Stream(Vec.fill(n)(Bits(32 bits)))` would be more than enough !
It's more for the scala/spinal way of writing it that I'm not comfortable with.)

Thanks

andreasWallner, 2024-01-18:
Ah sorry, mixed up `Vec(Fragment)` and `Fragment(Vec)`.

What I meant was something like:
```scala
case class MyBlackBoxSpinal(n: Int = 2) extends Component {
  case class MyBlackBox() extends BlackBox {
    addGeneric("dwCount", n)
    val sStream_payload_fragment = in port Bits(n * 32 bit)
    val sStream_valid = in port Bool()
    val sStream_ready = out port Bool()
    val sStream_payload_last = in port Bool()
  }
  val io = new Bundle {
    val sStream = slave port Stream(Fragment(Vec(Bits(32 bits), n)))
  }
  val inner = MyBlackBox()
  inner.sStream_valid := io.sStream.valid
  inner.sStream_payload_last := io.sStream.last
  io.sStream.ready := inner.sStream_ready
  inner.sStream_payload_fragment.subdivideIn(n slices) := io.sStream.payload.fragment
}
```
You can get the rework step to work for that as well - and that might be a good solution if you have a bunch of these.

KireinaHoro, 2024-01-22:
Take a look at [InOutVecToBits](https://github.com/SpinalHDL/SpinalHDL/blob/68b6158700fc2440ea7980406f927262c004faca/lib/src/main/scala/spinal/lib/io/InOutVecToBits.scala) that just got merged a bit earlier; the idea of how to use it is in #1258.  I see that you want a `Fragment(Vec)` instead of a `Vec(Fragment)`, but the idea should be the same...

oletf, 2024-01-22:
Thanks to both of you,
I'm gonna go with the simple wrapper, I think that's exactly what I was looking for !
Definitely keeping the `InOutVecToBits` in mind, might come in handy for more complex things.

closed 2024-01-22 by oletf