SpinalHDL/SpinalHDL #1284: Playing around spinal in scala console fails

soartang, 2024-01-17:
I'm learning spinal in windows.

I downloaded the latest SpinalTemplateSbt and msys2. I'm good with sbt compile and run are fine, but have problems playing around spinalhdl in console. Below is my build.sbt with JDK 17.0.81

ThisBuild / version := "1.0"
ThisBuild / scalaVersion := "2.12.18"
ThisBuild / organization := "org.example"

val spinalVersion = "1.10.0"
val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)
...

Below is what I typed in console. I tried msys and native windows cmd, both gave the same errors.

scala> import spinal.core._
import spinal.core._

scala> val a = Bits(8 bits)
java.lang.NullPointerException: Cannot invoke "spinal.core.GlobalData.nodeAreInferringWidth()" because the return value of "spinal.core.BitVector.globalData()" is null
  at spinal.core.BitVector.setWidth(BitVector.scala:163)
  at spinal.core.BitsFactory.Bits(Bits.scala:33)
  at spinal.core.BitsFactory.Bits$(Bits.scala:33)
  at spinal.core.package$.Bits(core.scala:35)
  ... 55 elided

Dolu1990, 2024-01-17:
Hi,

SpinalHDL can't realy be used in console mode in this way. You have to provide a toplevel component : 

SpinalVerilog(new MyTopLevel)

And have all your hardware statements in that toplevel herarchy

Reason is that SpinalHDL has a lot of hidden stuff happening under the hood which require some setup (done by the SpinalVerilog function)

andreasWallner, 2024-01-18:
FYI in case the `sbt` startup time is annoying you and that's the reason why you want to use the REPL: you can just start `sbt` once and repeatedly run `runMain`, etc. in there - that makes it much quicker to use. (sorry in case you already knew)

soartang, 2024-01-18:
> FYI in case the `sbt` startup time is annoying you and that's the reason why you want to use the REPL: you can just start `sbt` once and repeatedly run `runMain`, etc. in there - that makes it much quicker to use. (sorry in case you already knew)

Thanks for pointing out this. Personally I prefer the console way (works like python console which I use a lot). Though '~runMain' produces immediate results, it runs the whole code and need a lot of println to show the results. 

With console, it seems to be more newbie-friendly. For example, to find out the difference between 'd := a |<< c' and 'd := a << c', I can put them in console and find out the difference.

soartang, 2024-01-18:
> Hi,
>
> SpinalHDL can't realy be used in console mode in this way. You have to provide a toplevel component :
>
> SpinalVerilog(new MyTopLevel)
>
> And have all your hardware statements in that toplevel [...]

Go it. Assume I'd like to learn the difference of '##' and '@@', as a newbie, the preferable way to do so is to type the following code (Pardon me if it has errors):  
```
val a = U'8'h5'
val b = U'8'hA'
println( a @@ b )
println( a ## b)

```
Is there a way?

Dolu1990, 2024-01-18:
Personnaly, i'm using Intellij Idea and that template project : 
https://github.com/SpinalHDL/SpinalTemplateSbt

Else you can use sbt to generate some verilog, see the readme of that template project.

Else you can do some one liner as : 
SpinalVerilog(new Component{val a = U"8'h5"; val b = U"8'hA"}).printRtl()

But the thing is that you need your repl to have the SpinalHDL compiler plugin in. i never tried.

soartang, 2024-01-19:
> Personnaly, i'm using Intellij Idea and that template project : https://github.com/SpinalHDL/SpinalTemplateSbt
>
> Else you can use sbt to generate some verilog, see the readme of that template [...]

OK, i'll figure it out. Thank you Dolu.