SpinalHDL/SpinalHDL #1285: Why not directly check the corresponding bit in the implementation of One-Hot State Machine?

hhucchenyixiao, 2024-01-18:
The generated one-hot state machine implemented like this:
```
 case(1) // synthesis parallel_case
      (((stateReg) & enumDef_IDLE) == enumDef_IDLE) : begin
      end
```
Instead of implemented directly like this:

```
 case(1) // synthesis parallel_case
      (stateReg[0]) : begin
      end
```

Dolu1990, 2024-01-18:
Hi,

I personnaly "never" have hand written some verilog so i don't know soooo much about it, is the (stateReg[0]) : begin widely supported ? or specific to a verilog revision ?

hhucchenyixiao, 2024-01-18:
> Hi,
> 
> I personnaly "never" have hand written some verilog so i don't know soooo much about it, is the (stateReg[0]) : begin widely supported ? or specific to a verilog revision ?

This writing method does not use any special Verilog syntax, it simply utilizes the characteristic that one-hot encoding have only one bit as 1. I used this method to write a state machine mentioned in a tutorial as an example. 

```verilog
`timescale 1ns/1ps

module OneHotFsm (
    output reg  io_result,
    input  wire clk      ,
    input  wire rst_n
);
    localparam BOOT    = 4'b0001;
    localparam STATE_A = 4'b0010;
    localparam STATE_B = 4'b0100;
    localparam STATE_C = 4'b1000;

    localparam BOOT_ID   = 0;
    localparam STATE_A_ID = 1;
    localparam STATE_B_ID = 2;
    localparam STATE_C_ID = 3;

    reg [3:0] state_reg, state_next;
    reg [7:0] counter;

    wire state_b_on_entry;
    wire state_b_on_exit; 

    assign state_b_on_entry = !state_reg[STATE_B_ID] && state_next[STATE_B_ID];
    assign state_b_on_exit = state_reg[STATE_B_ID] && !state_next[STATE_B_ID];

    always @(*) begin
        (* parallel_case *)
        case(1'b1)
            state_reg[STATE_A_ID] : begin
                state_next = STATE_B;
            end
            state_reg[STATE_B_ID] : begin
                if (counter == 8'd4) begin
                    state_next = STATE_C;
                end else begin
                    state_next = state_reg;
                end
            end
            state_reg[STATE_C_ID] : begin
                state_next = STATE_A;
            end
            default : begin
                state_next = STATE_A;
            end
        endcase
    end

    always @(*) begin
        if (state_b_on_exit) begin
            io_result = 1'b1;
        end else begin
            io_result = 1'b0;
        end
    end

    always @(posedge clk or negedge rst_n) begin
        if (!rst_n) begin
            state_reg <= BOOT;
        end else begin
            state_reg <= state_next;
        end
    end

    always @(posedge clk or negedge rst_n) begin
        if (!rst_n) begin
            counter <= 8'd0;
        end else if (state_b_on_entry) begin
            counter <= 8'h0;
        end else if (state_reg[STATE_B_ID]) begin
            counter <= counter + 8'd1;
        end
    end

endmodule
```

commit 9cce86b3 (2024-01-22): fix #1285 verilog emit ohhot test using x[y] instead of bit mask
closed 2024-01-22 by Dolu1990