SpinalHDL/SpinalHDL #1286: RegIf doesn't support fields of SpinalEnum type

louiecaulfield, 2024-01-18:
When trying to create a field of for a SpinalEnum type, the creation fails at [RegInst.scala#142](https://github.com/SpinalHDL/SpinalHDL/blob/eb5a3a705ab25bfc6e236a51352f02934c22572b/lib/src/main/scala/spinal/lib/bus/regif/RegInst.scala#L142)

I'm not familiar enough (yet) with Spinal's internals to estimate the amount of work to implement this, but it would make my code much more beautiful if SpinalEnums are supported :-)

Dolu1990, 2024-01-22:
@jijingg

jijingg, 2024-02-04:
Sorry for the late reply. It shouldn't be difficult to support ENUM, but what I would like to ask is, in what scenario will regef be used in registers? Can you give me a real example， @louiecaulfield

PR #1300 (2024-02-04): Fix RegIf SpinalEnum
Fix #1286
add a match case for SpinalEnum when init the reg
closed 2024-02-04 by jijingg