SpinalHDL/SpinalHDL #1290: IMasterSlave type of Stream payload

H1MSK, 2024-01-19:
I found out that the Stream will call out/in function of payload when being set to master / slave, causing all payload signals to have the same direction.

The relevant code is here:

https://github.com/SpinalHDL/SpinalHDL/blob/eb5a3a705ab25bfc6e236a51352f02934c22572b/lib/src/main/scala/spinal/lib/Stream.scala#L61-L65

This implementation seems not to care if payload has IMasterSlave trait.

I'm new to this language (or to be more precise, I'm new to this industry), so I'm wondering if this is a bug or no one just write code like this. 

#223

Readon, 2024-01-28:
The definition of Stream is an unidirection communication facility, so that payload should be in one direction and the same as valid.

H1MSK, 2024-01-28:
Oh, I see. So if I have to implement some interface that has more than one signal controlled by the slave, it should not be a Stream by definition.
Thanks a lot for your help!

closed 2024-01-28 by H1MSK