SpinalHDL/SpinalHDL #1291: How can I use "dophase" to check some diy list?

Mido-sun, 2024-01-22:
some example plz  scala rookie!!!

Dolu1990, 2024-01-22:
Hi,

What is dophase ?
You mean how to add a custom transformation phase  in SpinalHDL ?

Mido-sun, 2024-01-22:
> Hi,
> 
> What is dophase ? You mean how to add a custom transformation phase in SpinalHDL ?

yes

Dolu1990, 2024-01-22:
Have you seen this hidden part of the doc : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html#exploring-the-datamodel

Mido-sun, 2024-01-22:
> Have you seen this hidden part of the doc : https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html#exploring-the-datamodel

No ,I will read it now . Thank You

yportne13, 2024-01-23:
a small demo: https://github.com/yportne13/SpinalMemBist/blob/master/src/scala/main/resize.scala

andreasWallner, 2024-02-04:
Since this is two weeks old I guess we can consider this answered.

closed 2024-02-04 by andreasWallner