SpinalHDL/SpinalHDL #1295: The wave reload question of VCS simulation

xie-1399, 2024-01-28:
HI !
    I meet a question when using the VCS simulation，Here is my question：
    using "shift + L" in the VCS can reload the wave. However, when I recompile the test , and I want to reload the wave in the VCS，the error occurred：

[image]

what should I do if I want to reload the wave in VCS after recompile the test? Thank You

Dolu1990, 2024-01-29:
@wswslzp (Let's me know if you don't want pings, i'm now just unpilling all the email i have and try to notify the poeple related to specific issues :))

wswslzp, 2024-02-03:
Sometime, Verdi cannot reload the waveform is expected since the KDB/waveform are removed and regenerated while Verdi is openning for some specific Verdi/VCS version.

xie-1399, 2024-02-22:
how to stop the KDB/waveform regenerated when starting the simulation each time?

wswslzp, 2024-03-20:
I think there's no way to prevent from regenerating KDB. SpinalHDL remove the sim workspace each time you recompile tests by default as I remembered.

closed 2024-03-21 by xie-1399