SpinalHDL/SpinalHDL #1296: Is there any way to add delay at assignment?

developfpga, 2024-01-29:
Hi

Is there any way to add delay at assignment? In verilog, I can add delay like below. How can I do it in Spinal?

for combinatorial logic:
```verilog
assign #(0.025*time) do = r;
```

for register
```verilog
q <= #1 d
```

I need it in my simulation job. For example, I need to adjust input and output signal latency.

Dolu1990, 2024-01-29:
Hi,
Currently, the only way you could would be through a blackbox.

But overall, i guess it could be added natively.

developfpga, 2024-01-30:
Thanks, waiting for offical support.

Dolu1990, 2024-01-31:
Hmm that may take some time, i'm realy underwater.

jijingg, 2024-02-04:
Adding delay "# 1" to the code is not a good idea, and some teams even explicitly prohibit it in their specifications .
If you have sampling alignment issues, you should solve them in the correct way in testbench

andreasWallner, 2024-02-04:
Do you need the delay in your testbench? If yes and if you use SpinalSim then you can there is `delayed`: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/threadLess.html#thread-less-api
I used this a few times to simulate sync delays, etc. on modules that did not include synchronizers but I wanted to test with the expected delay (and built it directly into the sim driver). If you'd want to use this with an existing driver that you don't want to modify then you can have a small wrapper for simulating with `simPublic` signals that are driven from the test.

I'm not sure I think its a good idea to add this to the generation capabilities of the core language as its not synthesizable.

closed 2024-05-27 by developfpga