SpinalHDL/SpinalHDL #1303: StreamFragmentWidthAdapter.make missing earlylast argument

louiecaulfield, 2024-02-06:
The earlylast argument is not available in the `StreamFragmentWidthAdapter`'s `make` methods

https://github.com/SpinalHDL/SpinalHDL/blob/2527c7c6b0fb0f95e5e1a5722a0be732b364ce43/lib/src/main/scala/spinal/lib/Stream.scala#L1833

louiecaulfield, 2024-02-07:
Sorry to open this issue again, but further implementing the width adapter, I noticed the dataMask is not available when using the `make` convenience methods as the Area, containing the datamask, is not returned.

I worked around it by defining my own `make` functions as below, but I'm not so satisfied by the style.

```Scala
class StreamFragmentWA
object StreamFragmentWA {
  def apply[T <: Data,T2 <: Data](input : Stream[Fragment[T]],output : Stream[Fragment[T2]], endianness : Endianness, earlyLast : Boolean = false): Bits = {
    StreamFragmentWA(input, output, endianness, false, earlyLast)
  }

  def apply[T <: Data,T2 <: Data](input : Stream[Fragment[T]],output : Stream[Fragment[T2]], endianness : Endianness, padding : Boolean, earlyLast : Boolean): Bits = {
    StreamFragmentWidthAdapter(input, output, endianness, padding, earlyLast).dataMask
  }

  def make[T <: Data, T2 <: Data](input : Stream[Fragment[T]], outputPayloadType : HardType[T2], endianness : Endianness, earlyLast : Boolean) : (Stream[Fragment[T2]], Bits) = {
    val ret = Stream(Fragment(outputPayloadType()))
    val mask = StreamFragmentWA(input,ret,endianness,false,earlyLast)
    (ret, mask)
  }
}
```

The  in my module I write
```Scala
val (axi_stream, axi_mask) = StreamFragmentWA.make(fifo.io.pop,
                                                    Bits(io.axi.config.dataWidth bits),
                                                    LITTLE,
                                                    earlyLast = true)
```

Dolu1990, 2024-02-12:
Hmm what's about adding it to StreamFragment, but with an alternative name ?
makeWithMask ?

(PR welcome) ^^

commit 48c8a108 (2024-02-07): Fix #1303