SpinalHDL/SpinalHDL #1304: How to generate RTL with parameters?

developfpga, 2024-02-07:
I want to generate an IP with parameters like 
``` verilog
module ip #(
parameter dataWidth = 4
)(
input [dataWidth-1:0] d_i,
output [dataWidth-1:0] q_o
);

assign q_o = d_i;

endmodule
```

I want the RTL generate by spinal with parameter reserved.

Dolu1990, 2024-02-07:
Hi,

This is impossible to have the generated netlist having verilog parameters. The reason is that SpinalHDL + scala are doing the hardware elaboration. 
For sure, there would be theoricaly some ways to support some verilog parameters, but it would hit limitations very fast. In other words, this would go against the SpinalHDL paradigm.

Depending your workflow, one solution could be to release a JVM jar file of your scala / SpinalHDL generator, so people could generate new version of the verilog using command line arguments.

developfpga, 2024-02-19:
Thanks，I got it

closed 2024-02-19 by developfpga