SpinalHDL/SpinalHDL #1307: StreamMux with streamed select no correct join

g0t00, 2024-02-08:
I have a problem with using Stream Mux with the `createSelector` helper when using this factoy
```scala
  def apply[T <: Data](select: Stream[UInt], inputs: Vec[Stream[T]]): Stream[T] = {
```
this is using this to generate the selector:
```scala
def createSelector(): Stream[UInt] = new Composite(this, "selector") {
      val stream = Stream(cloneOf(select))
      val reg = stream.haltWhen(input.isStall).toReg(U(0))
      select := reg
    }.stream
```
This construction breaks when the inputs stream gets valid before the select stream.
This does not get correctly tested for in mux_with_selector formal test, as there is no assert validating the select stream with the output. (Only the value after createSelector register)
I am currently using a workaround where I join the select stream with the output of the mux to fix the handshaking.
But I do not know if this is the cleanest solution, also this breaks the createSelector API. (But I am unsure if this is fixable without breaking that):
```scala
def apply[T <: Data](select: Stream[UInt], inputs: Seq[Stream[T]]): Stream[T] = {
    val c = new StreamMux(inputs(0).payload, inputs.length)
    (c.io.inputs, inputs).zipped.foreach(_ << _)
    c.io.select := select.payload
    StreamJoin(c.io.output, select).map(_._1)
  }
```
I will provide draft PR with this included, but unsure how to best procede...

likewise, 2024-02-08:
> This does not get correctly tested for in mux_with_selector formal test,
> as there is no assert validating the select stream with the output.
> (Only the value after createSelector register)
>
I would strongly suggest first to write a formal test that proves the bug.

Then it will also prove the fix.

And we are sure we are not introducing new bugs, and have a regression test against the bug re-occuring in later refactorings.

Especially the primitives like FIFOs and Muxes are important to have correct IMHO.

g0t00, 2024-02-08:
I fully agree with your statement that the primites should be correct for sure. But I am currently only learning how to work with formals and I do not really no how to correctly formulate a generic set of assertions that would correctly fail on the current solution and work on my current work around.
In the dev-branch version there is a register, my current workaround does not have one. I could make a classic TB that checks for that though...
@likewise The problem is in FormalMuxTester.scala line 117-118. This assertion does use dut.io.select for asserting correct muxing. It would instead need to use the muxSelector streams value for verification. Looking at other formal tests HistoryModifyable might be usable here but as there is zero documentation here I have a hard time wrapping my head around this. If you have more experience on this topic I would appreciate any help there.

Readon, 2024-02-18:
> I have a problem with using Stream Mux with the `createSelector` helper when using this factoy
>
> ```scala
>   def apply[T <: Data](select: Stream[UInt], inputs: Vec[Stream[T]]): Stream[T] = {
> ```
>
> this [...]

Which version are you using? In mainstream branch, the StreamMux's createSelector is using:
``` scala
val reg = stream.haltWhen(output.isStall).toReg(U(0))
```
but demux is. 
``` scala
val reg = stream.haltWhen(input.isStall).toReg(U(0))
```

This is done to ensure that the selection register is only changed if the output stream from StreamMux and the input stream from StreamDemux are not stalled.

The rationale for the formal test you mention is to check that the output is the same as the chosen input, subject to a finite number of constraints.

It does not check the equality and timing between the select reg and corresponding stream payload.

Would you please describe what have you met on your case with a minimum example? Thanks.

Readon, 2024-02-18:
I suppose what you want is a select stream synchronized with the output stream. 
The current design is only for asynchronized usage, you need to implement that yourself.

Dolu1990, 2024-02-19:
> I suppose what you want is a select stream synchronized with the output stream.
The current design is only for asynchronized usage, you need to implement that yourself.

I think asynchronized / syncronized usage are 2 possibilities of egual "rank", probably we should change the name of things to be explicite on "what you get"

ex . StreamMux.syncSel(s1, s2) StreamMux.regSel(s1, s2)  or something in that sort ?

Readon, 2024-02-19:
> I think asynchronized / syncronized usage are 2 possibilities of egual "rank", probably we should change the name of things to be explicite on "what you get"
>
> ex . StreamMux.syncSel(s1, s2) [...]

as normal definition, mux is a combinational concept, which is natrually asynchronized. also, it is different from stream dispatcher.

the synchronized using can be easily archieved by adding a synchronized stream fork to select stream and join one of output select streams with mux's output stream.

Dolu1990, 2024-02-19:
> as normal definition, mux is a combinational concept, which is natrually asynchronized. also, it is different from stream dispatcher.

I understand, but at the same time, there all the Stream stuff, which also lead to understand things on a transaction basis, not combinatorinal (that's why i would tend to not use the "object StreamMux's apply" function, but instead having more explicit function name. That way it doesn't rely anmore on implicit understanding.

g0t00, 2024-02-19:
> I understand, but at the same time, there all the Stream stuff, which also lead to understand things on a transaction basis, not combinatorinal (that's why i would tend to not use the "object [...]

I completely agree. I also like you comment with the two methods instead of using apply. I will provide an updated pr which does not change existing behavior.

Readon, 2024-02-19:
> I completely agree. I also like you comment with the two methods instead of using apply. I will provide an updated pr which does not change existing behavior.

Then, how about add the Demux version also?😊

PR #1308 (2024-02-28): StreamMux with stream select use join

Closes #1307 

# Context, Motivation & Description

Proposed fix for #1307.
Not sure how to handle createSelector. AFAIK it can not be fixed as the input stream has to be halted when there is no valid selection. Should we just remove?

# Impact on code generation

# Checklist

- [X] Unit tests were modified
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Docume [... cut]
closed 2024-02-28 by Readon