SpinalHDL/SpinalHDL #1310: WishboneSlaveFactory - doWrite is never called in pipelined mode

janschiefer, 2024-02-13:
It may be totally possible that I am completely and utterly wrong...

but I have been playing with WishboneSlaveFactory and and noticed some odd behaviour:

- When driving WishboneSlaveFactory from a master in "classic" mode everything works fine.

- However, when I try a "single write operation" in Wishbone B4 **pipelined mode** with a cycle according to specification, the data is never written to the register. In fact the internal signal "askWrite" is fired, but the signal "doWrite" is never activated.

[image]

Wishbone specification B4:
[image]

If I extend the STB-Pulse one more cycle - which is not according to the specification - "doWrite" is called and the register is successfully written:

[image]

The problem seems to be, that SpinalHDL Lib's wishbone implementation also expects STB to be high i**n the cycle when ACK is received**. However, the specification requires STB only to be high in the cycle before ACK when in pipelined mode:

[image]

My WB Slave test code:
```
package tangBusTest

import spinal.core._
import spinal.lib._
import spinal.lib.bus.wishbone._

// Hardware definition
case class WishboneLed(wb_config: WishboneConfig) extends Component {
  val io = new Bundle {
    val leds_out = out Bits (8 bits)
    val wb_bus = slave(Wishbone(wb_config))
  }

  val slave_factory = WishboneSlaveFactory(io.wb_bus)

  if (wb_config.useERR) io.wb_bus.ERR := False

  val leds_reg = slave_factory.createReadWrite(Bits(8 bits), address = 0x08, bitOffset = 0)

  val test_read_reg = slave_factory.read(U(value = 112, 8 bits), address = 0x16, bitOffset = 0)

  io.leds_out := leds_reg

}

object WishboneLed {}
```

I hope, I have found a valid issue and not wasted your time.

janschiefer, 2024-02-13:
Versions:
ThisBuild / scalaVersion := "2.12.18"
val spinalVersion = "1.10.1"
Java version:  OpenJDK Runtime Environment Temurin-17.0.10+7 (build 17.0.10+7)

andreasWallner, 2024-02-14:
Thanks for the extensive report!

I had a look at the Wishbone spec and I'd also say that this looks wrong - I'll have a bit more of a look tomorrow though because I've not looked at wishbone in a long time.
It looks to me like the `WishboneDriver`s `sendPipelinedBlockAsMaster` does the right thing though, we'll have to have a looks what going on there.

andreasWallner, 2024-02-18:
Yes, I can confirm that this is wrong, I was curious why our tests did not pick this up... looks to me like they were not finished back in 2018, they don't verify that the register was really written... 
If you want to open a PR you're welcome, otherwise I can have a look end of this week.