SpinalHDL/SpinalHDL #1322: NamedType returns null when called outside of a Component

KireinaHoro, 2024-02-21:
I'm trying to restructure some of my common code between components with Scala `trait`s, but it seems like we cannot create `NamedType`s for use with `HardMap` properly inside traits:

```scala
package pionic

import spinal.core._

case class Good() extends Component {
  val KeyA = NamedType(Bool())
  val hm = HardMap(Seq(KeyA))
}

trait BadTrait {
  this: Component =>

  val KeyB = NamedType(Bool())
  val hm = HardMap(Seq(KeyB))
}
case class Bad() extends Component with BadTrait

object NamedTypeNull extends App {
  SpinalVerilog(Good())
  SpinalVerilog(Bad())
}
```

Generates the following NPE:

```
Exception in thread "main" scala.MatchError: null (of class spinal.core.NamedType)
        at spinal.core.HardMap.keyToName(HardMap.scala:20)
        at spinal.core.HardMap.update(HardMap.scala:28)
        at spinal.core.HardMap.add(HardMap.scala:32)
        at spinal.core.HardMap$.$anonfun$apply$1(HardMap.scala:11)
        at spinal.core.HardMap$.$anonfun$apply$1$adapted(HardMap.scala:11)
        at scala.collection.immutable.List.foreach(List.scala:333)
        at spinal.core.HardMap$.apply(HardMap.scala:11)
        at pionic.BadTrait.$init$(NamedTypeNull.scala:14)
        at pionic.Bad.<init>(NamedTypeNull.scala:16)
        at pionic.NamedTypeNull$.$anonfun$new$2(NamedTypeNull.scala:20)
        at spinal.core.internals.PhaseCreateComponent.$anonfun$impl$273(Phase.scala:2556)
        at spinal.core.fiber.Engine$.$anonfun$create$1(AsyncCtrl.scala:147)
        at spinal.core.fiber.AsyncThread.$anonfun$jvmThread$1(AsyncThread.scala:59)
        at spinal.core.fiber.EngineContext.$anonfun$newJvmThread$1(AsyncCtrl.scala:39)
        at spinal.sim.JvmThread.run(SimManager.scala:51)
```

Is there a way to make this pattern work?

Dolu1990, 2024-02-26:
> trait BadTrait {
  this: Component =>

  val KeyB = NamedType(Bool())
  val hm = HardMap(Seq(KeyB))
}

Yes right, that is done on purpose to avoid valcalback being done on parent classes.
If you want to have the valcalback done on it, the usual way is to make your trait extends Area (Area is kinda the generalized named scope)

KireinaHoro, 2024-02-26:
> Yes right, that is done on purpose to avoid valcalback being done on parent classes. If you want to have the valcalback done on it, the usual way is to make your trait extends Area (Area is kinda the generalized named scope)

I get the point, but however this doesn't allow mix-ins:

```scala
package pionic

import spinal.core._

class MyBase extends Component {
  val o = out(Bool())
}

trait BadTrait extends Area {
  this: MyBase =>

  o := True
}
case class MyMod() extends MyBase with BadTrait

object NamedTypeNull extends App {
  SpinalVerilog(MyMod())
}
```

Gives the following error:

```
[error] Tester.scala:14:12: class MyMod inherits conflicting members:
[error]   def rework[T](gen: => T): T (defined in class Component) and
[error]   def rework[T](body: => T): T (defined in trait Area)
[error]   (note: this can be resolved by declaring an `override` in class MyMod.)
[error] case class MyMod() extends MyBase with BadTrait
```

KireinaHoro, 2024-02-26:
One general example is with [Chipyard](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Configs-Parameters-Mixins.html#cake-pattern-mixin) in Chisel.  You can mix in different fragments of hardware with freedom to compose a final SoC.

A smaller and more concrete example is to build composable smartNICs: you can have different:
- host interfaces (PCIe, normal AXI for Zynq, CXL, [ECI](https://unlimited.ethz.ch/display/enzianwiki/ECI+-+Enzian+Coherent+Interconnect), etc.)
- datapath components (TCP offloading, encryption accelerators, etc.)
- Ethernet MAC implementation (Intel, Xilinx, etc.)

And you can structure these as traits to mix-in and build your final design.  IMO something similar could probably be achieved with the [Plugin](https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html#plugin) system from VexiiRiscv, but I feel it's a bit too high caliber for smaller designs and it's not in `spinal.lib`.

KireinaHoro, 2024-02-27:
Hmmm I'll try to see if my design works with the plugin system

Dolu1990, 2024-02-28:
> One general example is with [Chipyard](https://chipyard.readthedocs.io/en/stable/Chipyard-Basics/Configs-Parameters-Mixins.html#cake-pattern-mixin) in Chisel. You can mix in different fragments of hardware with freedom to compose a final SoC.

Ahh i would say, it maybe a bad idea, overall, mixing trait feel very hacky to me. I know in chipyard they do it a lot, but i realy "feel" like this is wrong. I mean,  i don't realy see the point of doing it like that. as ineritence is normaly a tool made to provide software interfaces, not to compose a system ?

> And you can structure these as traits to mix-in and build your final design. IMO something similar could probably be achieved with the [Plugin](https://spinalhdl.github.io/VexiiRiscv-RTD/master/VexiiRiscv/Framework/index.html#plugin) system from VexiiRiscv, but I feel it's a bit too high caliber for smaller designs and it's not in spinal.libupdate: found [it](https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/misc/plugin/Fiber.scala).

Hmm, there is many feature that we should consider separatly.

- Composing a system
- Having the element of that system being able to discover each other and ask stuff to each other at elaboration time
- Scheduling the elaboration of the different parts of the system "dynamicaly"

The curse with 

```scala
class DigitalTop(implicit p: Parameters) extends ChipyardSystem
  with testchipip.tsi.CanHavePeripheryUARTTSI // Enables optional UART-based TSI transport
  with testchipip.boot.CanHavePeripheryCustomBootPin // Enables optional custom boot pin
  with testchipip.boot.CanHavePeripheryBootAddrReg // Use programmable boot address register
  with testchipip.cosim.CanHaveTraceIO // Enables optionally adding trace IO
  with testchipip.soc.CanHaveBankedScratchpad // Enables optionally adding a banked scratchpad
  with testchipip.iceblk.CanHavePeripheryBlockDevice // Enables optionally adding the block device
  with testchipip.serdes.CanHavePeripheryTLSerial // Enables optionally adding the backing memory and serial adapter
  with testchipip.soc.CanHavePeripheryChipIdPin // Enables optional pin to set chip id for multi-chip configs
  with sifive.blocks.devices.i2c.HasPeripheryI2C // Enables optionally adding the sifive I2C
  with sifive.blocks.devices.pwm.HasPeripheryPWM // Enables optionally adding the sifive PWM
  with sifive.blocks.devices.uart.HasPeripheryUART // Enables optionally adding the sifive UART
  with sifive.blocks.devices.gpio.HasPeripheryGPIO // Enables optionally adding the sifive GPIOs
  with sifive.blocks.devices.spi.HasPeripherySPIFlash // Enables optionally adding the sifive SPI flash controller
  with sifive.blocks.devices.spi.HasPeripherySPI // Enables optionally adding the sifive SPI port
  with icenet.CanHavePeripheryIceNIC // Enables optionally adding the IceNIC for FireSim
  with chipyard.example.CanHavePeripheryInitZero // Enables optionally adding the initzero example widget
  with chipyard.example.CanHavePeripheryGCD // Enables optionally adding the GCD example widget
  with chipyard.example.CanHavePeripheryStreamingFIR // Enables optionally adding the DSPTools FIR example widget
  with chipyard.example.CanHavePeripheryStreamingPassthrough // Enables optionally adding the DSPTools streaming-passthrough example widget
  with nvidia.blocks.dla.CanHavePeripheryNVDLA // Enables optionally having an NVDLA
  with chipyard.clocking.HasChipyardPRCI // Use Chipyard reset/clock distribution
  with chipyard.clocking.CanHaveClockTap // Enables optionally adding a clock tap output port
  with fftgenerator.CanHavePeripheryFFT // Enables optionally having an MMIO-based FFT block
  with constellation.soc.CanHaveGlobalNoC // Support instantiating a global NoC interconnect
{
  override lazy val module = new DigitalTopModule(this)
}
```

Is "what's about i need 2 HasPeripheryGPIO ?" It feel like the cake pattern can't handle it (but i maybe wrong) If it is right, it mean it isn't flexible enough to compose a system.

In terms of composing a SoC, have you seen : 
https://github.com/SpinalHDL/VexiiRiscv/blob/408761518cd332ac41321932243cbe1f7fb2447c/src/main/scala/vexiiriscv/soc/demo/MicroSoc.scala#L27

Note it doesn't use templates, that would be something to consider separatly, as you could see HasPeripheryGPIO as a template.

andreasWallner, 2024-03-04:
Another possible option how an SoC can be parameterized with peripherals:
https://github.com/andreasWallner/spinalStuff/blob/master/src/main/scala/andreasWallner/spinaltap/SpinalTap.scala#L197
(in that case it doesn't have a core, but the main idea remains)

I feel this is addressing a similar issue (and one could still pass a config for a core as well) but a bit differently. Could / would need to be combined with the fiber stuff if one wanted to be more flexible w.r.t. bus architecture though.

KireinaHoro, 2024-03-05:
Seems like for now Plugins fit my use-case quite good for my use case.  I'm tempted to keep this open if we want to come back to this...

andreasWallner, 2024-03-06:
As it currently stands I'd guess that chances are low that we implement the Chipyard stuff as is - let's close and reopen if you come across another concrete issue (as I understood you have a solution for your current problem)?