SpinalHDL/SpinalHDL #1326: AxiLite4SlaveFactory generates combinatorial paths from inputs to ready outputs

456Xander, 2024-02-23:
# The Problem

The AXI4-Lite specification forbids any combinatorial paths from an input to an output in slaves.
However, a AXI4-Lite slave generated with the AxiLite4SlaveFactory has multiple combinatorial paths for the ready signals.
* ar.ready is equal to r.ready when `r.valid` is true, a fixed true when `r.valid` is false
* aw.ready and `w.ready` (identical value) are just `aw.valid & w.valid`

## Simple reproducible example

```scala
package package

import spinal.core._
import spinal.lib._
import spinal.lib.bus.amba4.axilite.AxiLite4
import spinal.lib.bus.amba4.axilite.AxiLite4Config
import spinal.lib.bus.amba4.axilite.AxiLite4SlaveFactory

case class TestAXI() extends Component {
    val io = new Bundle {
        val bus = slave port AxiLite4(AxiLite4Config(8, 32))
        val bit = out port Bool()
    }
    val busCtrl = AxiLite4SlaveFactory(io.bus)
    val bit_r   = busCtrl.createReadWrite(Bool(), 0, 0)
    io.bit := bit_r
}

object TestStreamVhdl extends App {
    Config.spinal.generateVhdl(new TestAXI())
}
```

I synthesized the generated VHDL code in Quartus Prime 23.1, which results in the Hardware as shown in the picture below (Netlist viewer)

[image]

Here the combinatorial paths are marked in red.

## Working on a solution

All ready signals should come from registers.

I may attempt to write a correct SlaveFactory for AXI-Lite.
However I am not very familiar with either SpinalHDL nor AXI-Lite (I have mostly used the Intel Avalon Bus before), so it may take a while.
I will update this issue with any progress I make.

In the meantime, I would welcome any feedback about this issue or if there is already a solution being worked on.

Dolu1990, 2024-02-26:
Should be good with https://github.com/SpinalHDL/SpinalHDL/commit/d04f565baf76fbbeef745cc8d47cb613b1aa4909, let's me know if there is any issue left ^^

456Xander, 2024-02-26:
I don't have a practical issue with it right now, as the design I'm working on still early on and only runs in simulation, where I noticed the core not following spec.

As far as I understand, the reason for the no combinatorial paths rule in the spec is, because it may introduce long combinatorial paths is the design, that reduce max clock speed.

You're right that registering data and address with a halfPipe on w and aw uses unnecessary area, especially as the spec guarantees that data and address are stable as long as valid is high.

To follow spec, it would be enough to register the valid signals of both paths, a validPipe should be enough and only cost 2 registers.

I'm also OK with leaving it how it is right now, and testing it again once my design is more complete and see if it causes a practical issue.

Dolu1990, 2024-02-26:
> To follow spec, it would be enough to register the valid signals of both paths, a validPipe should be enough and only cost 2 registers.

I forgot that one existed XD. Yes it would work, as long as the BusSlaveFactory haltIt primitive isn't used by the user (as that one may depend on the address.

> I'm also OK with leaving it how it is right now, and testing it again once my design is more complete and see if it causes a practical issue.

In general, i tend to integegrate pipelining stages at a more "macro level", meaning between the main memory bus and the shared peripheral bus, which then goes to individual peripherals, that can save a lot of area, especialy if you have many little bus slave factory (ex gpio)

commit d04f565b (2024-03-04): Fix #1326