SpinalHDL/SpinalHDL #1327: Signal naming oversights

jjjt-git, 2024-02-24:
When building a module based component (so a component which gets some functionality injected via functions), any dash in filenames can be carried down to verilog, causing syntax errors (I will provide a demonstration of this behaviour, as soon as I have more time).

Another minor thing I noticed, when trying to hit specific naming schemas with AXI, is that there is (as far as I am aware) currently no way to suppress the underscore as structure-delimiter.
This would help in hitting (for the part after the bus-prefix) <channel><signal>, like they are named in AMBA4-Spec. However, this would be more of a QoL-feature.

Dolu1990, 2024-02-24:
Hi,

I'm not sure to understand the dash thing, can you provide an example ?

jjjt-git, 2024-02-24:
Still working on reproducing it in a minimal form. Dumping my entire project here would hardly help.
But, in case it helps, when renaming the file from "some-file.scala" to "SomeFile.scala", the error disappeared.

Dolu1990, 2024-02-24:
Ahhhhhhhhhhhhhhhhhhhhh, i got it now.
Yes right, that isn't filtred. the easyest way to fix that would be to filter in the spinal.idslplugin.Location class i think

commit 37d99371 (2024-02-26): Fix #1327 Add Location.fileSymbol