SpinalHDL/SpinalHDL #1330: FiberPlugin addPrePopTask

KireinaHoro, 2024-02-27:
To rename top level ports, I previously used `addPrePopTask` to traverse through `getAllIo` of the `Component` to rename them.  With plugins, this does not work on the top level module any more since the plugins are only resolved during build in fiber execution.  Seems like the following works:

```scala
class Top extends Component {
  val host = new PluginHost
}
object Top {
  def apply(plugins: Seq[Hostable]): Top = {
    val n = new Top
    n.host.asHostOf(plugins +: new FiberPlugin {
      during build {
        // rename IO here
      }
    }
  }
}
```

However, since to ensure order we need locking (via `Retainer`), it seems like there's no guarantee that this fiber plugin to rename the IO will actually execute in the very end.  Is there a way to ensure that a plugin executes at last (like `create late` in NaxRiscv)?

KireinaHoro, 2024-02-27:
A potential solution I see is to add a `postBuildLock` that is retained by default in `FiberPlugin`; `during build` releases this lock after reworking the body of build.  We then make the IO renaming plugin await on all of these post build locks to ensure that it runs last.

KireinaHoro, 2024-02-28:
I'm putting all IO generation in modules into `setup` and doing the manipulation in a plugin during build for now.  I do like the awaitPatch idea, since this would not require moving the IO definitions :)

Dolu1990, 2024-03-04:
Added in SpinalHDL composable-exp and dev branch

KireinaHoro, 2024-03-07:
Works! Closing.

closed 2024-03-07 by KireinaHoro