SpinalHDL/SpinalHDL #1335: When to override `clone`?

KireinaHoro, 2024-03-04:
I'm a bit confused on when should I override `clone` for a datatype.  For example:

```scala
case class TestUnion() extends Union {
  val a = newElement(UInt(4 bits))
  val b = newElement(Bits(4 bits))
}
```

If I don't override `clone`, calling clone on an instance of `TestUnion` and then accessing the member would result in a NPE:

```scala
val a = TestUnion()
val b = a.clone.asInstanceOf[TestUnion]

b.a := U(15) // <-- NPE at spinal.core.UnionElement.$anonfun$get$1(Union.scala:12)
```

But if I override clone as:

```scala
override def clone: TestUnion = TestUnion()
```

It works fine.  Is there a rule on when should I override clone?

Dolu1990, 2024-03-04:
Hi,
That's a bug i just fixed now.
You normaly need to implement clone only for Bundle which aren't case class and have parameters (either explicit or from the scala execution stack)

KireinaHoro, 2024-03-04:
The commit doesn't fix the NPE given by the example, when clone is not overriden...

KireinaHoro, 2024-03-04:
So the example I have is:

```scala
import spinal.core._

case class TestUnion() extends Union {
  val a = newElement(UInt(4 bits))
  val b = newElement(Bits(4 bits))
}

object TestStreamUnion extends App {
  SpinalVerilog(new Component {
    val a = TestUnion()
    val b = a.clone.asInstanceOf[TestUnion]
    b.a := U(3)
  })
}
```

Exception:

```[Runtime] SpinalHDL dev    git head : 5e40a78aa6a4b5cf8be503eed1f5b084101bbaab
[Runtime] JVM max memory : 16012.0MiB
[Runtime] Current date : 2024.03.04 13:44:47
[Progress] at 0.000 : Elaborate components

ERROR !
A null pointer access has been detected in the JVM.
This could happen when in your SpinalHDL description, you access an signal which is only defined further.
For instance :
  val result = Bool()
  result := a ^ b  //a and b can't be accessed there because they are only defined one line below (Software rule of execution order)
  val a,b = Bool()

Exception in thread "main" java.lang.NullPointerException
        at spinal.core.UnionElement.$anonfun$get$1(Union.scala:12)
        at scala.collection.mutable.HashMap.getOrElseUpdate(HashMap.scala:469)
        at spinal.core.signalCache$.apply(Misc.scala:169)
        at spinal.core.signalCache$.apply(Misc.scala:172)
        at spinal.core.UnionElement.get(Union.scala:12)
        at spinal.core.UnionElement$.wrapped(Union.scala:8)
        at pionic.TestStreamUnion$$anon$1.<init>(TestStreamUnion.scala:14)
        at pionic.TestStreamUnion$.$anonfun$new$1(TestStreamUnion.scala:11)
        at spinal.core.internals.PhaseCreateComponent.$anonfun$impl$273(Phase.scala:2556)
        at spinal.core.fiber.Engine$.$anonfun$create$1(AsyncCtrl.scala:147)
        at spinal.core.fiber.AsyncThread.$anonfun$jvmThread$1(AsyncThread.scala:59)
        at spinal.core.fiber.EngineContext.$anonfun$newJvmThread$1(AsyncCtrl.scala:39)
        at spinal.sim.JvmThread.run(SimManager.scala:51)
```

KireinaHoro, 2024-03-04:
I have no idea what that commit ID is; my local branch definitely includes bb24b25dba1364dbf4016798a448f19c3468d587 .

Dolu1990, 2024-03-04:
Ahh fixed, you were triggering another clone case escaping the init :D

closed 2024-03-04 by Dolu1990

andreasWallner, 2024-03-06:
Btw. as a small note: in usual case we'd recommend to use `case class` instead of `class` for `Bundle`s and `Components` which will auto-generate a correct `clone` routine.

commit bb24b25d (2024-03-04): Fix #1335
commit 27b71e4c (2024-03-04): Fix #1335 Data.clone now always purify the data