SpinalHDL/SpinalHDL #1337: `assignDontCare` on Bundle element inside Union does not work

KireinaHoro, 2024-03-04:
```scala
import spinal.core._
import spinal.lib._

case class TestBundle() extends Bundle {
  val xb5 = Bits(5 bits).assignDontCare()
  val u = UInt(32 bits)
}

case class TestUnion() extends Union {
  val b = newElement(TestBundle())
}

object TestStreamUnion extends App {
  SpinalVerilog(new Component {
    val a = TestUnion()
    val b = TestBundle()
    b.u := U(7, 32 bits)
    a.b.u := b.u
  })
}
```

Complains:

```
LATCH DETECTED from the combinatorial signal (toplevel/a :  Bits[37 bits]), unassigned bit mask is 00000_00000000000000000000000000011111, defined at
    pionic.TestStreamUnion$$anon$1.<init>(TestStreamUnion.scala:18)
    pionic.TestStreamUnion$.$anonfun$new$1(TestStreamUnion.scala:17)
    spinal.sim.JvmThread.run(SimManager.scala:51)
```

Dolu1990, 2024-03-04:
Hi,

>   val xb5 = Bits(5 bits).assignDontCare()

No assignement should be done inside a Bundle definition, Bundle is realy just for type definition with eventualy tag and directions

What's about : 

```scala
    a.b.xb5.assignDontCare()
    a.b.u := b.u
```
or 
```scala
    a.raw.assignDontCare()
    a.b.u := b.u
```

> Notice how for b it doesn't complain i.e. assignDontCare actually works.

Right it isn't realy a bug.
The reason is that the union host everything in one single Bits signal and then emulate the union elements, while the vanilla bundle keep things as they are defined (separate signals for each elements)

More that Spinal will ignore some linting on signals which have no purpose
If you do : 

```scala
    val b = out(TestBundle())
    b.u := U(7, 32 bits)    
```

You will get a 

```
NO DRIVER ON (toplevel/b_xb5 : out Bits[5 bits]), defined at
...
```    

But as in your original example, b_xb5 had no purpose, it is ignored (to not be too ennoying with lint)

KireinaHoro, 2024-03-04:
Ah I get it now.  But supporting such a use case would be useful, for example for defining message bitfields: https://github.com/KireinaHoro/spinal-blocks/blob/245a348e277b6aebc543a2c1d831dd8ba208f606/blocks/jsteward/blocks/eci/EciCmdDefs.scala#L77-L179

Some of the bitfields are defined by protocol to not have any meaning.  It'd be quite annoying if I'd have to assign dontcare to them manually.

Dolu1990, 2024-03-04:
What you can do is the following : 
```scala
case class EciVcCatMreq0to10() extends Bundle {
  val opcode = EciOpcode
  val xb4 = Bits(4 bits)
  val rreqId = EciId
  val dmask = EciDmask
  val ns = Bool()
  val xb3 = Bits(3 bits)
  val xb2 = Bits(2 bits)
  val address = EciAddress
  def dontCare(){
    xb2.assignDontCare()
    xb3.assignDontCare()
    xb4.assignDontCare()
  }
}
```

And later call the dontCare function

But overall, i think what you realy want is to have some hole in the data structure right ?
maybe we should add some proper support for that
Maybe a tag to add some post/pre padding ?

Dolu1990, 2024-03-04:
Probably, we should find a way to define blanks in bundles, that need to impact the asBits / assignFromBits / aliasAs functions

That would solve it cleanly

Dolu1990, 2024-03-04:
Something like : 

```scala
case class EciVcCatMreq0to10() extends Bundle {
  val opcode = EciOpcode
  blank(4 bits)
  val rreqId = EciId
  val dmask = EciDmask
  val ns = Bool()
  blank(3 bits)
  blank(2 bits)
  val address = EciAddress
}
```

Or/and maybe something more explicit, specifying which field is at which offset

mapping(opcode -> 0, rreqId -> 6, ...)

KireinaHoro, 2024-03-05:
I like the idea of having `blank(BitWidth)` better compared to the mapping function, since that seems a bit more error-prone and repetitive:

- we have to calculate offsets manually (with something like `lastOffset + lastSize`)
- what if two fields overlap?

andreasWallner, 2024-03-06:
I'm thinking about whether `PackedBundle` be a solution to this problem as it allows you to specify how members of a bundle are layed out.

KireinaHoro, 2024-03-07:
I took a closer look and think that this can work, but it's still quite tedious to write these packFrom clauses since that requires hand-calculation of the bit offsets.  How about introducing a `hole` command to create a hole?

Dolu1990, 2024-03-08:
The most convenent API would be something which mix implicit / explicit. With explicit commands to seek / skip stuff, and else things just get sequancial ? That would kinda support all the usages.

```scala
case class EciVcCatMreq0to10() extends Bundle {
  val opcode = EciOpcode
  blank(4 bits) //skip 4 bits
  val rreqId = EciId
  val dmask = EciDmask
  val ns = Bool()
  moveAt(33) //seek to bit 33
  val address = EciAddress
  val addressxxx = EciAddress
}
```

Dolu1990, 2024-03-08:
> I took a closer look and think that this can work, but it's still quite tedious to write these packFrom clauses since that requires hand-calculation of the bit offsets. How about introducing a hole command to create a hole?

No idea

KireinaHoro, 2024-03-08:
Do you want to implement this on Bundle or PackedBundle?

Dolu1990, 2024-03-15:
Fundamentaly, it maybe implemented directly in MultiData, maybe.

dokleina, 2024-06-16:
I'm late to this, but I took a pass at adding `skipTo` to PackedBundle: https://github.com/dokleina/SpinalHDL/commit/140f231d0801301acb7a41b592e301283aa9e0d6

dokleina, 2024-06-19:
Update: I misread this thread 🤦‍♂️ 

I've updated my fix to add `skipOver` instead of `skipTo`: https://github.com/dokleina/SpinalHDL/commit/5f60664e45b3f33e267e5a48f8f4ae8c730c277d

You'd use it like in [Dolu's comment](https://github.com/SpinalHDL/SpinalHDL/issues/1337#issuecomment-1977215608), but with `skipOver` instead of `blank`. Seeking to a position would just be using `packFrom` with the absolute bit position.

@KireinaHoro as a complete example, your class would look like this:
```scala
case class EciVcCatMreq0to10() extends PackedBundle {
  val opcode = EciOpcode
  skipOver(4 bits)
  val rreqId = EciId
  val dmask = EciDmask
  val ns = Bool()
  skipOver(3 bits)
  skipOver(2 bits)
  val address = EciAddress
}
```

It's worth noting that you _do not_ need to explicitly specify every field's bit position with `packTo`/`packFrom`. You may choose to, but PackedBundle will automatically pack starting at the next bit position.

KireinaHoro, 2024-06-20:
Looks nice!  Dolu mentioned implementing this in MultiData, but tbh this is enough for me.  I guess you can work on a PR?

PR #1460 (2024-06-21): Added skipOver method to PackedBundle

Closes #1337 

# Context, Motivation & Description

Adds the `skipOver` method to PackedBundle, which enables specification of gaps in packing without needing explicitly calculated bit positions (i.e., `packFrom` or `packTo`). 

# Impact on code generation

None

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%2 [... cut]
closed 2024-06-21 by Dolu1990