SpinalHDL/SpinalHDL #1340: Debug instructions executed twice

patstew, 2024-03-06:
It seems like if you raise debug_cmd_valid less than 6 cycles after the previous instruction, the second instruction will be executed twice. This seems to be somehow related to IBusSimplePlugin_injector_decodeInput_valid being high, or the Fetcher otherwise passing the instruction on twice, but I'm finding it difficult to follow exactly what's supposed to be happening in there.

Dolu1990, 2024-03-06:
Hi,

This is quite possible, as the interface assume things are very slow to come (JTAG), never tested fast rate of commands.

One question is, do you use the debug_cmd_ready  ?

patstew, 2024-03-06:
Yes, I'm using the ready/valid signals. It's a little odd in that it claims to be ready, until valid is high, then goes low combinatorially for a few cycles.

Sorry I've just noticed I raised this in the SpinalHDL rather than the vexriscv repo, my bad.

patstew, 2024-03-06:
Here are some possibly relevant signals when it's 'working' with a longer gap:
[image]

And when it's not:
[image]

The yellow marker is at an ADDI instruction that gets executed twice. It seems like the fetcher is up to somehting quite different, even though the CPU should be halted by the debugger at his point.