SpinalHDL/SpinalHDL #1343: Trying to understand simulator wait apis (newbie)

hMihaiDavid, 2024-03-10:
Hello, I am writing a simple testbench for a Timer component that was written following same spec as the timer in the examples.

I wrote the following test driver and got an expect waveform (using verilator sim backend with fst wave):

```
object MyTopLevelSim extends App {
  val compiled = Config.sim.compile {
			val dut = new Timer(width = 4)
			dut.counter.simPublic()
			dut
		}

	compiled.doSim{dut =>
		SimTimeout(1000)
		dut.clockDomain.forkStimulus(10)

		dut.io.tick  #= true
		dut.io.clear #= false
		dut.io.limit #= 7

		for (i <- 0 until 10) {
			dut.clockDomain.waitSampling()
			//println("AAAAAAAAAA");
		}
		
		dut.io.limit #= 9
		for (i <- 0 until 10) {
			dut.clockDomain.waitSampling()
			//println("AAAAAAAAAA");
		}

	}
}
```

[image]

I'm doing this because I am trying to get a better understanding of how the simulator threads work, especially when is the eval() step of the verilated dynlib called and how can I synchronize reads and writes to signals with this eval loop.

For example, in the testbench above, my (limited, maybe wrong) understanding is that dut.clockDomain.forkStimulus(10) call somehow sets the clock and reset signal during each iteration of the eval loop, then, in the main sim thread, those three input signals are set and the loop after it waits for 10 cycles to pass.

Is it possible that for example dut.io.limit #= 7 is executed after some clock cycles have already been evaled by the model and thus some default/spurious values have been used? Or is it the call to waitSampling() what triggers an entire eval step of the model and the model is not executed until that function is called?

Thanks.

Dolu1990, 2024-03-13:
> For example, in the testbench above, my (limited, maybe wrong) understanding is that dut.clockDomain.forkStimulus(10) call somehow sets the clock and reset signal during each iteration of the eval loop

Not necessarly each eval loop.
SpinalSim emulate time on the top of verilator. So if you have other things happening at other times, that forkStimulus isn't involved.

>  dut.io.limit #= 7 is executed after some clock cycles have already been evaled by the model 

dut.io.limit #= 7  isn't part of the model, it is software.
in SpinalSim, when you do a #= it will always be visible after the end of the current delta cycle and before the start of the next delta cycle

waitSampling() will wait until the clock sampling is detected. It doesn't trigger eval.

In other words, SpinalSim emulate time on the top of verilator by having a queue of callback to execute on given timestamp. that's mostly it ^^

andreasWallner, 2024-03-16:
@hMihaiDavid You gave Dolu's answer a thumbs up, did his response answer your question? In that case we could close the Issue. 

Two quick hints:
- In the simulation threads are not "real" threads, but run serially so you don't need any locking etc.
- It's not required, but if you have some question that's not a bug or something you can ask them in the [gitter](https://app.gitter.im/#/room/#SpinalHDL_SpinalHDL:gitter.im) chat - more eyes there that can help

hMihaiDavid, 2024-03-17:
It's much more clear now, thank you both.

closed 2024-03-17 by hMihaiDavid