SpinalHDL/SpinalHDL #1346: SimMutex locked twice would crash the simulation

KireinaHoro, 2024-03-11:
```scala
object TestMutex extends App {
  val sim = SimConfig.withConfig(SpinalConfig()).withVerilator.allOptimisation.withFstWave

  val dut = sim.compile(new Component {
    val o = master(Event)
    o.valid := True
  })
  dut.doSim("access-union-member") { dut =>
    dut.clockDomain.forkStimulus(period = 4)

    val m = SimMutex()

    StreamReadyRandomizer(dut.o, dut.clockDomain)
    StreamMonitor(dut.o, dut.clockDomain) { _ =>
      m.lock()
    }

    sleep(1000)
  }
}
```

Crashes the simulation thread:

```
Exception in thread "main" java.lang.NullPointerException
        at spinal.core.sim.package$SimMutex.lock(package.scala:1099)
        at pionic.TestUnionSim$.$anonfun$new$2(TestUnionSim.scala:27)
        at pionic.TestUnionSim$.$anonfun$new$2$adapted(TestUnionSim.scala:26)
        at spinal.lib.sim.StreamMonitor.$anonfun$new$2(Stream.scala:33)
        at spinal.lib.sim.StreamMonitor.$anonfun$new$2$adapted(Stream.scala:33)
        at scala.collection.IterableOnceOps.foreach(IterableOnce.scala:576)
        at scala.collection.IterableOnceOps.foreach$(IterableOnce.scala:574)
        at scala.collection.AbstractIterable.foreach(Iterable.scala:933)
        at spinal.lib.sim.StreamMonitor.$anonfun$new$1(Stream.scala:33)
        at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2(package.scala:1016)
        at spinal.core.sim.package$SimClockDomainPimper.$anonfun$onSamplings$2$adapted(package.scala:1016)
```

Looking at `SimMutex.lock`, it seems like if we repeatedly lock a mutex, at the third invocation we'd have invoked `simThread.suspend()` a second time, but `simThread.suspend` replaces `simThread` (aka `SimManagerContext.current.thread`) with `null`.  I imagine the thread from the `StreamMonitor` should be blocked already on the second `lock` and thus not invoke `lock` again... What's happening here?

Dolu1990, 2024-03-11:
yes
I would say it realy depend what you want to do overall. What are you trying to do ?

KireinaHoro, 2024-03-11:
I just realised I was abusing the mutex for my own book-keeping.  I'm going to think this through again so closing for now.

andreasWallner, 2024-03-16:
Just as a hint: all `fork`ed "threads" are greenthreads, there is no real multithreading there, so you generally don't need any synchronization and can store state just in a "normal" variable.