SpinalHDL/SpinalHDL #1349: JTAG VPI Simulation gets blocked by IO

rpls, 2024-03-12:
I recently switched my VexRiscv simulation to the official RISC-V JTAG and JTAG VPI as a network interface and was wondering why the simulation suddenly is excruciatingly slow. Turns out, that the JTAG VPI implementation has a blocking IO call in its main loop. Line 81, i.e., the `readFully` call is blocking. Since it's executed in a `SimThread` the simulation gets blocked. In practice, its not fully blocked, since a connected OpenOCD on the VPI port will send regular packages, so the simulation just becomes really slow.

https://github.com/SpinalHDL/SpinalHDL/blob/7c6b13792617daef52f6b706632925f71ebde99e/lib/src/main/scala/spinal/lib/com/jtag/sim/JtagVpi.scala#L79-L83

A trivial fix would be to do a `while` loop that waits until enough bytes are present in the input buffer, the more elegant fix would probably to do some proper asynchronous IO (e.g., incremental non-blocking reading of `VpiCmd` packages).

Dolu1990, 2024-03-15:
Hi ^^

For the PR; let's me know when you think it isn't draft :)

Note, i also recently added https://github.com/SpinalHDL/SpinalHDL/blob/composable-exp/lib/src/main/scala/spinal/lib/com/jtag/sim/JtagRemote.scala#L23
Which is in regular openocd

rpls, 2024-03-16:
> For the PR; let's me know when you think it isn't draft :)

I wanted to run some more tests. Played around with a network-thread (Proper thread, not SimThread), that does all the networking, parsing etc. and just passes the VpiCmds.... turns out to be a lot slower. But I'm no Java/Scala Networking Programmer, so I might just have messed it up. Anyway, let's leave it as-is.

>Note, i also recently added https://github.com/SpinalHDL/SpinalHDL/blob/composable-exp/lib/src/main/scala/spinal/lib/com/jtag/sim/JtagRemote.scala#L23
>Which is in regular openocd

Also a nice option! I'll test which is faster for my use-case.

PR #1350 (2024-05-13): jtagvpi: Don't use blocking IO, to avoid blocking the simulation

Closes #1349 

# Context, Motivation & Description

A first trivial fix for the blocking IO problem in the JTAG VPI implementation. Just adds a `while` loop with a `sleep`, which is similar to the behaviour of the `JtagTCP` implementation.

# Impact on code generation

Shouldn't have any impact on code generation, just the simulation. 

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks t [... cut]
closed 2024-05-13 by Dolu1990