SpinalHDL/SpinalHDL #1353: Dump ports of `Mem` in FST waveforms?

KireinaHoro, 2024-03-13:
I noticed that in a SpinalSim simulation with Verilator backend, the `Mem` instances are not dumped in the resulting FST waveform.  While this might be necessary due to performance reasons, it seems like the access ports of the RAM are generated with the `_zz` prefix and thus not dumped (or not displayed?) in GTKWave.  Is there a way to make the ports accessible in the waveforms?

Dolu1990, 2024-03-15:
Hi,

For the Mem itself, it is a verilator setting, you can use the SimConfig.addSimulatorFlag("xxx") to change the verilator behaviour.
For the _zz access port not being displayed, yes, that's also a verilator behaviour to not record what start with a underscore.

Maybe the hardware generation should compose a name from the Mem name itself + a postfix for those, that would be cleaner, so a ComponentEmitterVerilog upgrade.
Else, you can change the anonymous prefix : 
SimConfig.withConfig(SpinalConfig(anonymSignalPrefix = "zz"))....

Dolu1990, 2024-03-18:
Should be fixed now

commit c8871a1c (2024-03-18): fix #1353
commit 071f579b (2024-03-18): fix #1353
closed 2024-03-18 by Dolu1990