SpinalHDL/SpinalHDL #1372: Blackboxing RAM with one read and one readWrite port

KireinaHoro, 2024-03-21:
I understand that for Mem, a readWrite port can always host a read or a write port, so in principle the RAM with 1 readSync and 1 readWrite port should be able to be blackboxed into a `Ram_2wrs`.  This is not possible currently:

```
class spinal.core.blackboxAll$ is not able to blackbox toplevel/bindProtoToCoreCtrl_txPktBuffer : Mem[151*512 bits]                                                                          
  write ports : 0                                                                                                                                                                            
  readAsync ports : 0                                                                                                                                                                        
  readSync ports : 1                                                                                                                                                                         
  readWrite ports : 1                                                                                                                                                                        
  -> Unblackboxable memory topology 
```

KireinaHoro, 2024-03-21:
I think in practice RAM like this would be possible to infer by the EDA tool.  I'm currently using the blackbox flow to make sure I don't accidentally create RAM that could not be inferred by Vivado; do we have some sort of checks for this purpose?

Dolu1990, 2024-03-22:
>  should be able to be blackboxed into a Ram_2wrs

But shoudnl't as the idea is to be very accurate about what it is (Ram_1r_1wrs which doesn't exists)

> I think in practice RAM like this would be possible to infer by the EDA tool,

Ahh on the big vendors yes, but on smaller things / darker place, things can break very easily
For vivado, i only use blackboxing when using byte enable or multiple write ports

> do we have some sort of checks for this purpose?

See https://github.com/SpinalHDL/VexiiRiscv/blob/litex/src/main/scala/vexiiriscv/soc/litex/Soc.scala#L295 for instance only ask blackboxing on byte enable rams