SpinalHDL/SpinalHDL #1376: Performance of PriorityMux

Readon, 2024-03-29:
While rethink on the #1359 under ultra wide input scenario, the PriorityEncoder is required, which request an efficient PriorityMux first. 

The existing PriorityMux generate the code in daisy chain mode, as below.
``` Verilog
output_1 <= (sel[0] ? inputs_0 : (sel[1] ? inputs_1 : (sel[2] ? inputs_2 : inputs_3)));
```
This heavily depends on the capability of the synthesizer to reduce it's critical path.

Does anyone have investigated on this pattern with various tools? If this is necessary to be refactored?

Dolu1990, 2024-03-29:
> The existing PriorityMux

Which one is it in the code base ? can't find it ?

personnaly i'm doing : 

```scala
OhMux.or(OHMasking.firstV2(xxx))
```

Readon, 2024-03-29:
[Here](https://github.com/SpinalHDL/SpinalHDL/blob/10340f035ed35bb05c258efb0d87e1ada40693de/lib/src/main/scala/spinal/lib/Utils.scala#L1343) is the link. 

I only find one usage of it is sFindFirst.

Dolu1990, 2024-03-29:
Ahhh, don't know how i missed it XD

Yes that is pretty bad. Could compose a better alternative from other utilities (ex OhMux.or(OHMasking.firstV2(xxx)))

Readon, 2024-03-29:
I just tried to add one. but I would try OhMux.or(OHMasking.firstV2(xxx)) first.

Dolu1990, 2024-03-29:
No idea what is best at the end, your implementation may win ^^

Readon, 2024-03-29:
I use yosys compare those two

```
=== PriorityMuxH16 === //OhMux.or(OHMasking.firstV2(xxx)) version

   Number of wires:                142
   Number of wire bits:            653
   Number of public wires:          84
   Number of public wire bits:     375
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                311
     $_AND_                         15
     $_DFF_P_                        8
     $_MUX_                        128
     $_NOT_                         15
     $_OR_                         145

=== PriorityMuxH16 === // direct version.

   Number of wires:                 57
   Number of wire bits:            296
   Number of public wires:          35
   Number of public wire bits:     169
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                139
     $_DFF_P_                        8
     $_MUX_                        120
     $_OR_                          11
```

Dolu1990, 2024-03-29:
Nice :D

Readon, 2024-03-29:
This the 1024 bit comparison.

=== PriorityMuxH1024 ===//OhMux.or(OHMasking.firstV2(xxx)) version

   Number of wires:               9841
   Number of wire bits:          42941
   Number of public wires:        7137
   Number of public wire bits:   39684
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              21098
     $_AND_                       1023
     $_DFF_P_                        8
     $_MUX_                       8192
     $_NOT_                       1023
     $_OR_                       10852

=== PriorityMuxH1024 ===// direct version.

   Number of wires:               3585
   Number of wire bits:          18944
   Number of public wires:        3098
   Number of public wire bits:   18352
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               9205
     $_DFF_P_                        8
     $_MUX_                       8184
     $_OR_                        1013

andreasWallner, 2024-04-02:
Which yosys settings are you using? I'd like to benchmark some ideas that I wrote a short while back to compare :-)

Readon, 2024-04-02:
These are commands I tested with yosys.
``` yosys
read -sv PriorityMuxH1024.v
hierarchy -top PriorityMuxH1024
proc; opt; techmap; opt
stat
```

Dolu1990, 2024-04-02:
Note, i'm suprised that OhMux.or(OHMasking.firstV2(xxx)) version infer $MUX 8192

The place where i could see yosys doing that would be "sel ? value.asBits | B(0, widthOf(value) bits))"

But in reality that would be just a AND gate.
So, i'm not sure where, but maybe one optimization pass of yosys was skiped ?

Readon, 2024-04-02:
so  if only and is adopted, theresults would not change that much.

Readon, 2024-04-03:
> But in reality that would be just a AND gate. So, i'm not sure where, but maybe one optimization pass of yosys was skiped ?

I have tested it with opt -full option, where only AND gate is adopted, the results would not change that much.

```
=== PriorityMuxH1024 ===

   Number of wires:               9841
   Number of wire bits:          42941
   Number of public wires:        7137
   Number of public wire bits:   39684
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              21098
     $_AND_                       9215
     $_DFF_P_                        8
     $_NOT_                       1023
     $_OR_                       10852
```

Dolu1990, 2024-04-03:
Hmm hard to tell, and/or gate could be half the area of mux (per gate)
https://diychip.org/sky130/sky130_fd_sc_ms/cells/and2/
https://diychip.org/sky130/sky130_fd_sc_ms/cells/mux2/
Even more if the synthesis is capable to use nand / nor instead, would be 33% the area per gate
https://diychip.org/sky130/sky130_fd_sc_ms/cells/nand2/

Also, OhMux.or is likely good/better in terms of area when you have wide busses. ex 32bits

Maybe, both way have some different sweet spots.

Readon, 2024-04-03:
I am still thinking about split the fundamental modules like priority mux/encoder/decoder like thing to another layer which is fit only to specific platform. ASIC or FPGA, even different vendors.

andreasWallner, 2024-04-03:
@Readon I started thinking about the same thing, especially for ASIC vs. FPGA where there is something to be gained from replacing some of the basic building blocks.
Major example would be small FIFOs which on the FPGA we'll often implement via a memory, for ASIC often an implementation with FFs that build a delay line is a bunch more efficient. (depending on the size). The same I can see for e.g. a bunch of other components.
Example that I recently looked at to some degree: priority encoders that have quite different sweetspots depending on width/technology to the degree that you can't (yet?) depend on the synthesis tool like here: https://www.semanticscholar.org/reader/14dc299ef6b9d15b285987d6fae5b941f6a017e8

Not sure though how valid it is to compare without a technology as a basis...

Readon, 2024-04-04:
so if we can allow overriding the initiated object by objects of its subclass, then we can keep the existing architecture unchanged. 

Just map vendor specific thing to general one. 
Class level or object level, is a problem. Class level replacment is convenient. Sometimes, we just need to replace only small group of object.

Need to investigated how scala helps to do that.

Readon, 2024-04-07:
> @Readon I started thinking about the same thing, especially for ASIC vs. FPGA where there is something to be gained from replacing some of the basic building blocks. Major example would be small [...]

I also read this article and preparing to implement the priority encoder as they do.

andreasWallner, 2024-04-09:
> I also read this article and preparing to implement the priority encoder as they do.

I can share my implementation of it in the evening.

PR #1377 (2024-04-25): Priority mux

Closes #1376

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?
closed 2024-04-25 by Dolu1990