SpinalHDL/SpinalHDL #1379: Redundant Flow API Calls.

MahirAbbas, 2024-03-30:
Redundant functions. Clutters up API.
Specifically in this case:
in Flow
```
x.stage() 
```
is equivalent to
```
x.m2sPipe()
```

```
def stage() : Flow[T] = this.m2sPipe()
```
x.stage() is undocumented.

Request: either remove one of the redundant definitions or document both

MahirAbbas, 2024-03-30:
```m2sPipe``` is also more descriptive while ```.stage()``` is not descriptive and is somewhat opaque. Redundant functions make codebases more difficult to read.

Dolu1990, 2024-03-30:
It depend the perspective, different language are used.
In the context of an interconnect, m2sPipe may make sense, because you want to specificaly cut combinatorial path around 
In the context of an pipeline, stage may make more sense, because "stage"
at this point, i would say documenting both would be the way to go ?

Dolu1990, 2024-03-30:
Yes thanks :)

closed 2024-03-30 by Dolu1990