SpinalHDL/SpinalHDL #1380: Parallel test execution in SpinalSim?

KireinaHoro, 2024-04-05:
My test suite is growing larger and single tests are starting to run longer.  Using the Verilator backend, I tried to issue multiple `dut.doSim` in parallel with [Scala parallel collections](https://github.com/scala/scala-parallel-collections), but it seems like only one simulation instance would progress at a time (despite using N CPU cores at 100% for N instances).  Is there a possibility to run multiple tests (on the same `SimCompiled[T]`) in parallel?

andreasWallner, 2024-04-09:
If you just want to speed up your regression then I'd recommend to just configure `testForkedParallel in Test := true` in your `build.sbt`, this way multiple test classes will be executed in parallel by scalatest (if you use scalatest). I'd guess this would be your best bet if you have small to medium designs and just want to speed up your whole regression (and only works if you use scalatest)

Yes it should be possible to run multiple verilator instances in parallel but especially for long simulations I've seen diminishing results since verilator itself also parallelizes. If you run them on the same `SimCompiled` you'll get multiple instances of the test class instantiated. Those should then be running in parallel.
Did you try to compare the difference between running in parallel vs. running serially?

If you have large designs then it would also be an option to look into parallel execution on the verilator side (needs to be passed during compilation - not sure whether we have a flag for that)

Dolu1990, 2024-04-09:
Hi,

As far i used it, i didn't had issue forking multiple threads running different tests from the same SimCompiled instance.

How [Scala parallel collections](https://github.com/scala/scala-parallel-collections) was used ? (code sample)

KireinaHoro, 2024-04-09:
@andreasWallner thanks for the idea of going with ScalaTest!  I'm using mill so not sure how to do `testForkedParallel` as in sbt.  I'm asking the folks at mill on Gitter to see.

@Dolu1990 I didn't commit the parallelized test harness (which was before I ported to ScalaTest), so I don't have a full example.  The idea is that I collect all calls of `dut.doSim(_)`, put the lambda into a list `tests` , and did something like this:

```scala
val tests: mutable.ArrayBuffer[MyDut => Unit] = ???
val compiled: SimCompiled[MyDut] = ???
tests.toIndexedSeq.par.foreach { t =>
  compiled.doSim(t)
}
```

Dolu1990, 2024-04-09:
Ahh that is weird. Maybe the thread afinity issue or something like that

Can you try with something like : 

```scala
  val threads = for(t <- tests) yield {
    val thread = new Thread {
      override def run {
        compiled.soSim(t)
      }
    }
    thread.start
    thread
  }

  threads.foreach(_.join())

```

KireinaHoro, 2024-04-09:
@Dolu1990 that works even worse, I don't even get the simulation logs created in parallel.  Anyways I think I have a solution already, thanks for the help :)

closed 2024-04-09 by KireinaHoro