SpinalHDL/SpinalHDL #1382: Are there some bugs in clockDomain.waitSampling() API?

yuqi-ali, 2024-04-07:


yuqi-ali, 2024-04-07:
@Dolu1990

andreasWallner, 2024-04-09:
To answer your question directly: maybe, but none that we know of ATM.

One question that I'd have is: what is driving the clock in the CD? Is it driven from `forkStimulus?`

`waitSampling` is used in the library verification itself and in many other places, we've yet to see an issue. Without code to reproduce it on a freely available simulator we'll not really be able to help much.
(if you want to compare your implementation to the StreamMonitor in lib: https://github.com/SpinalHDL/SpinalHDL/blob/aa2e7d77b74a46c284cc262e0692450810b2240d/lib/src/main/scala/spinal/lib/sim/Stream.scala#L15)

Dolu1990, 2024-04-09:
> clockDomain.waitSampling() in the StreamMonitor 

This is weird, there hasn't been a waitSampling since quite long in the StreamMonitor, which version of SpinalHDL are you using ?

Else, can you send a minimal self contained example for us to reproduce ? (verilator is the reference implementation of SpinalSim)

yuqi-ali, 2024-04-10:
SpinalHDL Version:1.10.1
Scala:2.11.12
The clockDomain is driven by forkStimulus.
For my design, I have a dozen cases for the same simulation environment, and only one of them has such a problem(either Verilator or VCS), causing the simulation to fail.

yuqi-ali, 2024-04-10:
I have two stream monitor in my simualtion enviroment,one is ok.
the core base class is this :
https://github.com/xiaochuang-lxc/SpinalHDL-TB/blob/main/src/main/scala/tb/monitor/StreamSink.scala

zyn810039594, 2024-04-14:
So, maybe this's the same as my problem?[https://github.com/SpinalHDL/SpinalHDL/issues/1334](url)
In my issue, the problem is the clockDomain.onSampling API.

Readon, 2024-04-14:
This discussion has been last long, I am trying to find the difference between StreamMonitor and class you describe.

waitSampling is different from the onSampling, it waits specified number of valid samplings. onSampling triggered each valid sampling.
So some user's calling on waitSampling might interference to the one in the Monitor's main loop.
That's why onSampling is now used in the StreamMonitor design.

Another difference is that StreamMonitor is not to create a new process by fork. It uses the same process of simulation by default.

yuqi-ali, 2024-04-17:
Get Some Error Info:
The Failed Case has a lot of fork thread(dma queue,each queue has a fork thread)，when test 96 fork threads, the case passed.The case failed when fork 100 threads

andreasWallner, 2024-04-20:
Again: please post a self contained example that reproduces the issue - if not we can't help you.