SpinalHDL/SpinalHDL #1387: Unknown reason for hierarchy violation

tristanitschner, 2024-04-17:
Hi there,

so recently I decided to learn some Scala such that I can do hardware with SpinalHdl. However, I have been dumbfounded by a rather simple challenge: Instantiate an ICAPE2. Here is the code:

```scala
package ticomponents

import spinal.core._
import spinal.lib._

case class ICAPE2() extends BlackBox {

  addGeneric("ICAP_WIDTH", "X32")

  val io = new Bundle {
    val CLK   = in  Bool()
    val CSIB  = in  Bool()
    val RDWRB = in  Bool()
    val I     = in  Bits(32 bits)
    val O     = out Bits(32 bits)
  }

  mapClockDomain(clock=io.CLK)

}

case class TopLevel() extends Component {

  val io = new Bundle {
    val CSIB  = in  Bool()
    val RDWRB = in  Bool()
    val I     = in  Bits(32 bits)
    val O     = out Bits(32 bits)
  }

  val logic = new Area {
    val primitive = ICAPE2()

    primitive.io.CSIB  := io.CSIB
    primitive.io.RDWRB := io.RDWRB
    primitive.io.I     := io.I
    io.O               := primitive.io.O
  }

}

object ICAPE2VhdlGen extends App {
  SpinalVhdl(TopLevel())
}
```

And here's the relevant part of the error message:
```
[info] running (fork) ticomponents.ICAPE2VhdlGen 
[info] [Runtime] SpinalHDL dev    git head : 768f0960f581c3160c4ef14ebb03928f1f9e2db5
[info] [Runtime] JVM max memory : 15912.0MiB
[info] [Runtime] Current date : 2024.04.17 23:12:37
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 0.298 : Checks and transforms
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (5 errors).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.347 : Elaborate components
[info] [Progress] at 0.357 : Checks and transforms
[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  Error detected in phase PhaseCheckHierarchy
[error] ********************************************************************************
[error] ********************************************************************************
[error] HIERARCHY VIOLATION : (toplevel/[ICAPE2]/??? : in Bool) is driven by (toplevel/??? : in Bool), but isn't accessible in the null component.
[error]     ticomponents.TopLevel$$anon$3.<init>(icape2.scala:34)
[error]     ticomponents.TopLevel.<init>(icape2.scala:31)
[error]     ticomponents.ICAPE2VhdlGen$.$anonfun$new$1(icape2.scala:43)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
[error] HIERARCHY VIOLATION : (toplevel/[ICAPE2]/??? : in Bool) is driven by (toplevel/??? : in Bool), but isn't accessible in the null component.
[error]     ticomponents.TopLevel$$anon$3.<init>(icape2.scala:35)
[error]     ticomponents.TopLevel.<init>(icape2.scala:31)
[error]     ticomponents.ICAPE2VhdlGen$.$anonfun$new$1(icape2.scala:43)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
[error] HIERARCHY VIOLATION : (toplevel/[ICAPE2]/??? : in Bits[32 bits]) is driven by (toplevel/??? : in Bits[32 bits]), but isn't accessible in the null component.
[error]     ticomponents.TopLevel$$anon$3.<init>(icape2.scala:36)
[error]     ticomponents.TopLevel.<init>(icape2.scala:31)
[error]     ticomponents.ICAPE2VhdlGen$.$anonfun$new$1(icape2.scala:43)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
[error] HIERARCHY VIOLATION : (toplevel/??? : out Bits[32 bits]) is driven by (toplevel/[ICAPE2]/??? : out Bits[32 bits]), but isn't accessible in the null component.
[error]     ticomponents.TopLevel$$anon$3.<init>(icape2.scala:37)
[error]     ticomponents.TopLevel.<init>(icape2.scala:31)
[error]     ticomponents.ICAPE2VhdlGen$.$anonfun$new$1(icape2.scala:43)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] ********************************************************************************
[error] ********************************************************************************
[error] Design's errors are listed above.
[error] SpinalHDL compiler exit stack : 
[error]     at spinal.core.SpinalExit$.apply(Misc.scala:455)
[error]     at spinal.core.SpinalError$.apply(Misc.scala:510)
[error]     at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:177)
[error]     at spinal.core.internals.PhaseContext.doPhase(Phase.scala:193)
[error]     at spinal.core.internals.SpinalVhdlBoot$.$anonfun$singleShot$6(Phase.scala:2803)
[error]     at spinal.core.internals.SpinalVhdlBoot$.$anonfun$singleShot$6$adapted(Phase.scala:2801)
[error]     at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
[error]     at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
[error]     at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
[error]     at spinal.core.internals.SpinalVhdlBoot$.$anonfun$singleShot$1(Phase.scala:2801)
[error]     at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:71)
[error]     at spinal.core.internals.SpinalVhdlBoot$.singleShot(Phase.scala:2736)
[error]     at spinal.core.internals.SpinalVhdlBoot$.apply(Phase.scala:2731)
[error]     at spinal.core.Spinal$.apply(Spinal.scala:415)
[error]     at spinal.core.SpinalConfig.generate(Spinal.scala:180)
[error]     at spinal.core.SpinalVhdl$.apply(Spinal.scala:429)
[error]     at ticomponents.ICAPE2VhdlGen$.delayedEndpoint$ticomponents$ICAPE2VhdlGen$1(icape2.scala:43)
[error]     at ticomponents.ICAPE2VhdlGen$delayedInit$body.apply(icape2.scala:42)
[error]     at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error]     at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error]     at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error]     at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error]     at scala.collection.immutable.List.foreach(List.scala:431)
[error]     at scala.App.main(App.scala:80)
[error]     at scala.App.main$(App.scala:78)
[error]     at ticomponents.ICAPE2VhdlGen$.main(icape2.scala:42)
[error]     at ticomponents.ICAPE2VhdlGen.main(icape2.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 7 s, completed Apr 17, 2024, 11:12:38 PM
```

Am I doing something wrong or is this a bug?

fayalalebrun, 2024-04-20:
This example works for me. Which version of SpinalHDL are you using?

tristanitschner, 2024-04-20:
Ok, so it's working given the default template. However, what I was trying to do, was build against my local SpinalHdl source. Here is the build.sbt:

```scala
val spinalVersion = "dev"

lazy val spinalHdlCore       = ProjectRef(file("ext/SpinalHDL"), "core")
lazy val spinalHdlLib        = ProjectRef(file("ext/SpinalHDL"), "lib")
lazy val spinalHdlSim        = ProjectRef(file("ext/SpinalHDL"), "sim")

lazy val projectname = (project in file("."))
.settings(
        inThisBuild(List(
                version      := "1.0",
                scalaVersion := "2.12.18",
                organization := "org.example"
                )),
        Compile / scalaSource := baseDirectory.value / "hw" / "spinal"
        )
.dependsOn(spinalHdlSim, spinalHdlCore, spinalHdlLib)

fork := true

```

This is basically identical to how it's done in NaxRiscv Litex integration. However, I still can't get it to work. ext/SpinalHdl contains the latest dev branch.

fayalalebrun, 2024-04-20:
Looks like you might be missing the idsl plugin. Here is mine:
```
ThisBuild / version := "1.0"
ThisBuild / scalaVersion := "2.12.18"
ThisBuild / organization := "org.example"

lazy val spinalHdlIdslPlugin = ProjectRef(file("../SpinalHDL"), "idslplugin")
lazy val spinalHdlSim = ProjectRef(file("../SpinalHDL"), "sim")
lazy val spinalHdlCore = ProjectRef(file("../SpinalHDL"), "core")
lazy val spinalHdlLib = ProjectRef(file("../SpinalHDL"), "lib")

lazy val projectname = (project in file("."))
  .settings(
    Compile / scalaSource := baseDirectory.value / "hw" / "spinal",
  ).dependsOn(spinalHdlIdslPlugin, spinalHdlSim, spinalHdlCore, spinalHdlLib)

scalacOptions += (spinalHdlIdslPlugin / Compile / packageBin / artifactPath).map { file =>
  s"-Xplugin:${file.getAbsolutePath}"
}.value

fork := true
```

tristanitschner, 2024-04-20:
Yeah, that fixes it. Thank you very much.

closed 2024-04-20 by tristanitschner