SpinalHDL/SpinalHDL #1391: Slower on the intranet network

balanx, 2024-04-21:
os is win10
sbt 1.7.3
scala 2.12.18
java 11.0.19
SpinalHdl is 1.10.1

First, I compiled and run SpinalTemplateSbt  at the PC connecting to Internet. It's OK.
```
[Runtime] SpinalHDL v1.10.1    git head : 2527c7c6b0fb0f95e5e1a5722a0be732b364ce43
[Runtime] JVM max memory : 4078.0MiB
[Runtime] Current date : 2024.04.21 21:40:31
[Progress] at 0.000 : Elaborate components
[Progress] at 1.698 : Checks and transforms
[Progress] at 1.949 : Generate Verilog
[Done] at 2.243
```
Then, I only copied whole running environment to the other PC on the intranet, including "./sbt, ./ivy2" etc. 
SpinalTemplateSbt  can be run OK too, but there was one difference as the following,
```
[Progress] at 113.412 : Generate Verilog
[Done] at 1.918
```
The step "Generate Verilog" is very very slower !
What is the possible reason for this ?

Dolu1990, 2024-04-22:
Hi,

Hmm, it being so slow is realy weird.
I do not see a good reason for it to be related to the network itself.
In general when things get that slow on a given machine, it is more related to the amount of memory allocated to the JVM
What is reported by the : 
[Runtime] JVM max memory : ????

balanx, 2024-04-22:
Thanks for your reply.
Mm, I'll try ...

balanx, 2024-04-28:
I tried .
It will be normal after I turn off network at the "win10 / setting / network / ethernet / local connection" of the intranet PC.
Turn on again , the problem is reproduced.

andreasWallner, 2024-04-28:
Have you checked if there is some other tool that consumes more CPU when your network is turned off?
SpinalHDL is not accessing the network, so this seems really weird...

Also: Did you check whether the JVM memory config is the same if there is no network, like @Dolu1990 asked?

balanx, 2024-04-29:
> Have you checked if there is some other tool that consumes more CPU when your network is turned off? SpinalHDL is not accessing the network, so this seems really weird...
> 
> Also: Did you check whether the JVM memory config is the same if there is no network, like @Dolu1990 asked? Please post the full output of the fast/slow generation so that we can see where the time is spent.

The occupancy rate is very low regardless of whether the network is turned on or off.
The JVM memory config is the same regardless of whether there is internet.

I am just descripting how I have tried to solve the problem.
Again, I doubted sbt and  tried to use mill instead. The result is as follows.

[image]

Readon, 2024-04-30:
I think it's necessary to run a profiler to capture what's the primary time-consuming parts.

balanx, 2024-05-01:
@Readon may you give me a instruction for profier ?
@Dolu1990  I will try when I have the conditions.

andreasWallner, 2024-05-05:
@balanx if you are willing to to debug this, these instructions worked for me (though I'm on linux, so some details will be different for sure):
- https://medium.com/@barudwale20/demystifying-performance-profiling-in-java-scala-unleashing-the-power-of-hprof-for-memory-and-cpu-9ecf2a36a47b 
combined with https://github.com/cykl/hprof2flamegraph

But: considering the huge slowdown, it might be just as feasible to find what's the slow part by inserting print statements with the time/delta and doing binary search manually - might be much simpler then setting up the profiler.

Readon, 2024-06-20:
Nice, let's close this issue?

closed 2024-06-30 by andreasWallner