SpinalHDL/SpinalHDL #1392: Support for memories with read latency higher than 1

fayalalebrun, 2024-04-22:
At the moment, the `readSync` method of `Mem` produces a result in the following cycle (A read latency of 1). However, some memory IPs (see [here](https://github.com/KireinaHoro/spinal-blocks/blob/master/blocks/jsteward/blocks/mem/xpm/MemBlackbox.scala)) support read latencies higher than 2. Especially on FPGA's, this can be important to achieve high frequency.

Ideally, `Mem` would allow read latency to be configured, and convey this information to the blackboxing step.

Dolu1990, 2024-04-22:
Agree.

One question i have, how those 2 cycle latency memory IP support a clock enable for the second stage ? (if at all)

KireinaHoro, 2024-04-22:
[XPM] does not support a clock enable for the memory itself, only for the last register stage for output:

regcea	Input	1	clka	LEVEL_HIGH	1	Clock Enable for the last register stage on the output data path.
regceb	Input	1	clkb	LEVEL_HIGH	1	Clock Enable for the last register stage on the output data path.

[XPM]: https://docs.amd.com/r/en-US/ug974-vivado-ultrascale-libraries/XPM_MEMORY_TDPRAM

Dolu1990, 2024-04-22:
The way i see it to more foward would change the current : 

```scala
def readSyncImpl(address: UInt, data : Data, enable: Bool = null, readUnderWrite: ReadUnderWritePolicy = dontCare, clockCrossing: Boolean = false, allowMixedWidth: Boolean = false): Unit = {
```

into 

```scala
def readSyncImpl(address: UInt, data : Data, enable: Bool = null, readUnderWrite: ReadUnderWritePolicy = dontCare, clockCrossing: Boolean = false, allowMixedWidth: Boolean = false, latency : Int = 1, stageEnable : Seq[Bool] = Nil): Unit = {
```

And the adjuste things around it.

Same for the readWrite ports

That would fit ?

KireinaHoro, 2024-04-22:
Hmm I'm a bit confused about the stage enable signals being separate; what's the point of having them all separate?  For instance I'm not sure how will this map to the en/regce scheme of the XPM.

In a broader sense I also don't understand, why would you want to read the value from RAM but not enabling the output registers.

Dolu1990, 2024-04-22:
The point is to be able to implement streamed memory read of multiple cycle latency, with backpresure and the ability to collapse bubbles.

KireinaHoro, 2024-04-22:
That's indeed interesting.  I don't have a detailed idea how that would work, but from the viewpoint of the XPM, `stageEnable.head` should map to `ena`/`enb`, and `stageEnable.tail.reduce(_ && _)` should map to `regcea`/`regceb`?

Dolu1990, 2024-04-22:
regcea => enable
ena => stageEnable(0)

If that was a 3 cycle latency => 
ena_stage3 => stageEnable(1)

So in other words : 

Stage 1 => enable
Stage 2 => stageEnable(0)
Stage 3 => stageEnable(1)
and so on.

KireinaHoro, 2024-04-22:
No but this is not possible, since both en and regce are both single bit signals in the XPM, regardless of how many reg stages there are.  Or did I understand it wrong?

Dolu1990, 2024-04-22:
i would say it is possible up to 2 cycle latency for XMP.
Not sure there is any interrest using XMP with latency larger than 2 anyway "Values larger than 2 synthesize additional flip-flops that are not retimed into memory primitives."

KireinaHoro, 2024-04-23:
Sounds great.  I'm not too familiar with the internals of the Mem API, so it would be great if @Dolu1990 you can implement the fix; I can start upstreaming the XPM blackboxing pass I have as seen on Gitter :)

Dolu1990, 2024-04-25:
Will do.
Keep in mind, often in FPGA, unless you are very very high fmax, you don't want to use the output flipflop of the ramblock, instead you want those flops in the FPGA fabric which allows them to travel closer to where their output is used.

KireinaHoro, 2024-04-25:
Yeah that makes sense.  I think this should be left to the black boxing pass to handle, to use the output flop on the RAM primitive or to use separate stages?

Dolu1990, 2024-04-25:
I don't know.
Another idea would be that the blackboxing phase would be able to reconize Reg after the ram to fuse them into the blackbox XD

KireinaHoro, 2024-04-29:
Ah regarding the extra pipeline stages after reading [this thread on the Xilinx forum](https://support.xilinx.com/s/question/0D52E00007G0oFjSAJ/route-353-design-is-not-routable-as-its-congestion-level-is-6?language=en_US): they're useful to reduce routing congestion on the FPGA.  Pipelining with extra flops allow the placer to move logic away from the congestion site, which in turn improves QoR.

Dolu1990, 2024-04-30:
Hmm i'm think now that maybe a way to implement it would be to keep MemReadSync as it is, and add regular Reg on its read data, then we can add special SpinalTag to help inferation / blackboxing figureing out that those register can be merged.

That would have the advantage to keep things "simple" and not requiring any update in the verilog/vhdl backends.
This will onyl require the blackboxer to know about those special tag.

KireinaHoro, 2024-04-30:
That sounds exciting, but how would you guarantee that the unregistered RAM output is not consumed by anyone other than the reg stage?

Dolu1990, 2024-04-30:
What do you mean by consumed ?

developfpga, 2024-05-24:
Hi @KireinaHoro @Dolu1990 

I am interesting in this issue too.

Let's check xilinx ug974 as example
[image]
```scala
def readSyncImpl(address: UInt, data : Data, enable: Bool = null, readUnderWrite: ReadUnderWritePolicy = dontCare, clockCrossing: Boolean = false, allowMixedWidth: Boolean = false, latency : Int = 1, stageEnable : Seq[Bool] = Nil): Unit = {
```
enb is mapped to enable like before. And I think how about map stageEnable(0) to regceb?

Or we can use setBlackBoxName to generate ram blackbox, just add one more port 'stageEnable(0) '.

Dolu1990, 2024-05-24:
Hi @developfpga 

I did some experiments localy, by extending the readSyncImpl API, but now i think the best would be instead that the blackboxing phase be capable to reconize Reg after a readSync and fusion them into the blackbox.

That way, we could define pipeline in a regular way, and as long as there is a simple register after the read sync, all can be "magicaly" merged.

So what would be needed, is to implement a phase which iterate over the AST and add tags to Mem to document "fusion" possibilities, which can then be used by the blackboxer.

developfpga, 2024-05-24:
Hi @Dolu1990 

From my perspective and xilinx user guide, here is my advice.
```scala
readData := ram.readSync(addr, rdEn, latency, stageEnable)
```
default latency = 1, and no need stageEnable. and stageEnable.getBitsWidth == latency - 1
For example, if i write spinal like
```scala
readData := ram.readSync(addr, rdEn, 2, stageEnable)
```
The verilog code generated I expected is like 
```verilog 
always @(posedge rd_clk) begin
    if (rdEn) begin
      read_data <= ram[addr];
    end
    if (stageEnable) begin
        read_data_stage_1 <= read_data ;
    end 
end 
assign readData = read_data_stage_1 ;
```
The verilog blackbox is like, parameter **rdLatency** and io **rd_stage_enable** are added.
rdLatency and rd_stage_enable is handled by blackbox, no more to do by spinal.
```verilog
  blackbox #(
    .wordCount(1024),
    .wordWidth(36),
    .clockCrossing(1'b0),
    .technology("auto"),
    .readUnderWrite("dontCare"),
    .wrAddressWidth(10),
    .wrDataWidth(36),
    .wrMaskWidth(1),
    .wrMaskEnable(1'b0),
    .rdAddressWidth(10),
    .rdDataWidth(36),
    .rdLatency(2) // added
  ) u_ram(
    .wr_clk  (), //i
    .wr_en   (), //i
    .wr_mask (), //i
    .wr_addr (), //i
    .wr_data (), //i
    .rd_clk  (), //i
    .rd_en   (), //i
    .rd_addr (), //i
    .rd_data (),  //o
    .rd_stage_enable() // added
  );
```
So I don't understand about 'add tags', I don't think any tags need to be add here.

developfpga, 2024-05-30:
Hi @Dolu1990 

Any updates for this issue?

Dolu1990, 2024-05-30:
No progress so far, i'm quite under water :/

Dolu1990, 2024-05-31:
With the commit above, the following code seems fine : 

```scala
object PlayMemSync3 extends App{
  val config = SpinalConfig()
  config.addTransformationPhase(new MemReadBufferPhase)
  config.addStandardMemBlackboxing(blackboxAllWhatsYouCan)
  config.generateVerilog(new Component{
    val dataType = HardType(Rgb(5,6,5))
    val ram = Mem.fill(256)(dataType)
    val write = ram.writePort().asSlave
    val addr = in(ram.addressType)
    val readed = ram.readSync(addr)
    val miaou = dataType()
    miaou := readed
    val condA = in Bool()
    val buffer = RegNextWhen(miaou, condA)
    val result = out(CombInit(buffer))
  })
}
```

kartikp4892, 2024-06-01:
> With the commit above, the following code seems fine :
>
> ```scala
> object PlayMemSync3 extends App{
>   val config = SpinalConfig()
>   config.addTransformationPhase(new MemReadBufferPhase) [...]

This doesn't seem to be working with following slightly modified code where `buffer` is direct output of the module.

```
object PlayMemSync3 extends App{
  val config = SpinalConfig()
  config.addTransformationPhase(new MemReadBufferPhase)
  config.addStandardMemBlackboxing(blackboxAllWhatsYouCan)
  config.generateVerilog(new Component{
    val dataType = HardType(Rgb(5,6,5))
    val ram = Mem.fill(256)(dataType)
    val write = ram.writePort()
    write.asSlave
    val addr = in(ram.addressType)
    val readed = ram.readSync(addr)
    val miaou = dataType()
    miaou := readed
    val condA = in Bool()
    val buffer = RegNextWhen(miaou, condA)
    buffer.asOutput
    //val result = out(CombInit(buffer))
  })
}

```

Dolu1990, 2024-06-10:
@KireinaHoro  See https://github.com/SpinalHDL/SpinalHDL/issues/1392#issuecomment-2142160165
Let's me know if that works for you, also maybe this could be added as an option in the SpinalConfig

KireinaHoro, 2024-06-10:
@Dolu1990 I took a quick look, seems like the current API is that the blackbox pass infers automatically what's the maximum possible buffer stages to be absorbed into the RAM?  I think this works great, but sometimes I might want to *ensure* there's at least 2 cycles of latency (timing, etc.) and have elaboration fail if it's not possible.  For example:

```scala
object TestMemSyncLat extends App {
  val config = SpinalConfig()
  config.addTransformationPhase(new MemReadBufferPhase)
  config.addStandardMemBlackboxing(blackboxAllWhatsYouCan)
  config.generateVerilog(new Component {
    val dataType = HardType(Rgb(5, 6, 5))
    val ram = Mem.fill(256)(dataType)
    val write = ram.writePort().asSlave()
    val addr = in(ram.addressType)
    val data = ram.readSync(addr, latency = 2)  // <-- try to enforce a latency of 2 cycles
    val directOut = out(data.clone())
    directOut := data  // <-- but not possible due to read of intermediate data, should abort?
    val miaou = CombInit(data)
    val condA = in(Bool())
    val buffer = RegNextWhen(miaou, condA)
    val result = out(CombInit(buffer))
  })
}
```

KireinaHoro, 2024-06-10:
Another example of possible confusion is multiple independent buffers:

```scala
object TestMemSyncLat extends App {
  val config = SpinalConfig()
  config.addTransformationPhase(new MemReadBufferPhase)
  config.addStandardMemBlackboxing(blackboxAllWhatsYouCan)
  config.generateVerilog(new Component {
    val dataType = HardType(Rgb(5, 6, 5))
    val ram = Mem.fill(256)(dataType)
    val write = ram.writePort().asSlave()
    val addr = in(ram.addressType)
    val data = ram.readSync(addr)
    val miaou = CombInit(data)
    val condA = in(Bool())
    val buffer = RegNextWhen(miaou, condA)
    val result = out(CombInit(buffer))
    val buffer2 = RegNextWhen(miaou, condA)
    val result2 = out(CombInit(buffer2))
  })
}
```

The current implementation does not combine them and absorb it into the RAM.  I'm actually not sure what should be the ideal behaviour.  Personally I prefer in addition to the automatic absorption transformation phase, maybe also have a way to make the buffer intention explicit (and fail early).

Dolu1990, 2024-06-13:
@KireinaHoro Should be good now.
As long as the register assignement is in a single layer of when statement, and that the condition signal is exactly the same one, it should be all good.

Pyromuffin, 2025-05-31:
The rd latency infer feature works with a single stage, but for URAM primitives we want a latency > 2 to add pipelining registers. Right now I have to hardcode the rdLatency in the Ram_1w_1rs verilog file which is not good for supporting different different sized cascades. This is not for a fifo application so the read_ce signal isn't important, and in any case, URAM doesn't appear to support a clock enable for those internal registers (good one xilinx!) making fifo implementation a hassle anyway.