SpinalHDL/SpinalHDL #1406: Seed does not work at parallel test of spinal sim

developfpga, 2024-05-06:
Hi,

I got problems at parallel test of spinal sim.

I add configure to build.sbt
```scala
fork := true
testForkedParallel := true
```

And doing test like:
```scala

case class BitsConnect() extends Component {
	val io = new Bundle {
		val bitOut = out(Bits(3 bits))

		val bitIn = in(Bits(3 bits))
	}
	io.bitOut := io.bitIn
}

class RandomTester() extends AnyFunSuite with LazyLogging {

	object SelfSimConfig {
		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
	}

	var compiled: SimCompiled[BitsConnect] = null
	test("compiled") {
		compiled = SelfSimConfig().compile {
			val dut = new BitsConnect()
			dut
		}
	}

	test("bringup") {
		compiled.doSim("bringup", seed = 1) { dut =>
			dut.clockDomain.forkStimulus(10 ns)
			for (i <- 0 until 4) {
				val n = Random.nextInt(8)

				dut.io.bitIn #= n
				sleep(1)

				println("test 0 ", dut.io.bitOut.toBigInt)
				sleep(1)
				Thread.sleep(1000)
			}
		}
	}
}
class RandomTester1() extends AnyFunSuite with LazyLogging {

	object SelfSimConfig {
		def apply() = SimConfig.withWave.withConfig(SpinalConfig(targetDirectory = "gen",
			defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))).workspacePath("sim")
	}

	var compiled: SimCompiled[BitsConnect] = null
	test("compiled") {
		compiled = SelfSimConfig().compile {
			val dut = new BitsConnect()
			dut
		}
	}

	test("bringup") {
		compiled.doSim("bringup", seed = 0) { dut =>
			dut.clockDomain.forkStimulus(10 ns)
			for (i <- 0 until 4) {
				val n = Random.nextInt(8)

				dut.io.bitIn #= n
				sleep(1)

				println("test 1 ", dut.io.bitOut.toBigInt)
				sleep(1)
				Thread.sleep(1000)
			}
		}
	}
}
```
RandomTester and RandomTester1 are the same expected different seed.
The problem happens when execute 
```shell
sbt "test:testOnly *.RandomTester*"
```
The results are different each time I execute sbt test like
```txt
[Progress] Verilator compilation done in 2063.783 ms
[info] - compiled
[info] RandomTester1:
[Progress] Start BitsConnect bringup simulation with seed 1
(test 0 ,5)
[Progress] Verilator compilation done in 2615.391 ms
[info] - compiled
[Progress] Start BitsConnect bringup simulation with seed 0
(test 1 ,7)
(test 0 ,2)
(test 1 ,7)
(test 0 ,5)
(test 1 ,1)
(test 0 ,0)
(test 1 ,4)
[Done] Simulation done in 4198.806 ms
[info] - bringup
[info] - bringup
[Done] Simulation done in 4010.039 ms
```
or like 
```txt
[Progress] Verilator compilation done in 1460.317 ms
[info] - compiled
[info] RandomTester:
[Progress] Start BitsConnect bringup simulation with seed 0
(test 1 ,7)
[Progress] Verilator compilation done in 1806.722 ms
[info] - compiled
[Progress] Start BitsConnect bringup simulation with seed 1
(test 0 ,5)
(test 1 ,0)
(test 0 ,3)
(test 1 ,4)
(test 0 ,3)
(test 1 ,3)
(test 0 ,2)
[info] - bringup
[Done] Simulation done in 4046.391 ms
[Done] Simulation done in 4012.119 ms
[info] - bringup
```

But when I execute RandomTester alone, the output logs are always the same, like
```txt
[Progress] Verilator compilation done in 914.656 ms

[Progress] Start BitsConnect bringup simulation with seed 1

(test 0 ,5)
(test 0 ,0)
(test 0 ,3)
(test 0 ,4)
[Done] Simulation done in 4056.867 ms
```

So how can I use seed to reproduce the simulation results when doing parallel test?

Thanks.

Dolu1990, 2024-05-06:
Hi,

The issue seems to be Random.nextInt(8)

So, in Scala, Random is shared between all threads.

That's is why, in recent version on SpinalHDL, the import spinal.core.sim._ will bring you a simRandom that you should use instead of Random.

let's me know if the issue pressiste.

developfpga, 2024-05-10:
Hi, @Dolu1990 

Sorry for later reply, but it dose take a few days for us to port to latest version of SpinalHDL.

We are still trying

Dolu1990, 2024-05-13:
one main change is that if you use the SpinalHDL dev branch, the default Scala version is now 2.12.18

Dolu1990, 2024-05-13:
Also, let's me know if you have specific issues porting

developfpga, 2024-06-07:
Hi @Dolu1990 

sinRandom really works

Thanks

closed 2024-06-07 by developfpga