SpinalHDL/SpinalHDL #1407: Underscore in literals

robotman2412, 2024-05-06:
Other HDLs like Verilog let you put an underscore in literals, like so: `32'hffff_ffff`.
SpinalHDL does not allow this.

robotman2412, 2024-05-06:
NVM I put that in a masked literal, not a bits literal

closed 2024-05-06 by robotman2412

andreasWallner, 2024-05-10:
SpinalHDL does support that, for some examples see the first code block here: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Data%20types/bits.html#declaration

robotman2412, 2024-05-10:
> SpinalHDL does support that, also for bits literals. For some examples see the first code block here: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Data%20types/bits.html#declaration

Yeah but not for masked literals, which is what I was trying to do at the time

andreasWallner, 2024-05-11:
Ah - I understood that the wrong way around...

Dolu1990, 2024-05-13:
Hmm i just tried : 
    val x = U(42, 8 bits) === M"10--_0-01"
    
Which seems fine