SpinalHDL/SpinalHDL #1408: Linux: Spaces in path break formal verification

TheImunityGamer, 2024-05-06:
I installed SpinalHDL following the guide on the documentation. The Windows section mentions to avoid spaces in the project path so I assumed that on Linux it would be fine. Everything I've tested so far works other than formal verification. In `Project/./simWorkspace/unamed/unamed_bmc/ERROR` I am seeing
```
ERROR: sby file syntax error: '[files]' section entry expects up to 2 arguments, 9 specified
```
I am guessing that this is because in the `config.sby`, there are spaces in the [files] section. The fragments of the path that are separated by spaces are then treated as separate arguments, which is why there are 9 arguments. I tried adding \ before all the spaces but when I used runMain again, the modified config just got overridden.

Readon, 2024-05-08:
> I did some experimenting and could not get `sby` to accept spaces, neither by using quotes or by escaping with . Therefore it seems like this is an issue with [sby](https://github.com/YosysHQ/sby) and should be reported there.
> 
> By the way, if you want to test modifications to the `.sby` file, you can invoke the `sby` command directly.

add quotes is a reasonable solution, which might solve problems both on Windows and Linux. PR is welcome, or I will have a look at it later.

fayalalebrun, 2024-05-08:
> add quotes is a reasonable solution, which might solve problems both on Windows and Linux. PR is welcome, or I will have a look at it later.

Adding quotes did not work when I tried it, so I think either sby has some other way of specifying file paths with spaces, or they don't support it.

TheImunityGamer, 2024-05-09:
I got it working by using `../rtl/unamed_1.sv`. Or at least I got past that stage. Now i'm getting
```
sby unamed_1_bmc                                                                                                                                                                       ─╯
SBY 18:36:24 [unamed_1_bmc] engine_0: smtbmc --progress yices
SBY 18:36:24 [unamed_1_bmc] base: starting process "cd unamed_1_bmc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 18:36:24 [unamed_1_bmc] base: bash: line 1: cd: unamed_1_bmc/src: No such file or directory
SBY 18:36:24 [unamed_1_bmc] base: Can't open log file `../model/design.log' for writing!
SBY 18:36:24 [unamed_1_bmc] base: finished (returncode=1)
SBY 18:36:24 [unamed_1_bmc] base: task failed. ERROR.
SBY 18:36:24 [unamed_1_bmc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:36:24 [unamed_1_bmc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 18:36:24 [unamed_1_bmc] summary: engine_0 (smtbmc --progress yices) did not return a status
SBY 18:36:24 [unamed_1_bmc] summary: engine_0 did not produce any traces
SBY 18:36:24 [unamed_1_bmc] DONE (ERROR, rc=16)
```
I don't think that sby has a way of specifying file paths with spaces and I think you need to use relative paths. I also think that under the hood SpinalHDL does something more complex than running sby because sby didn't do it.

Readon, 2024-05-09:
so what's your purpose that to include space in path? it seems that sby itself does not support this well.

can you provide a minimal example first?

TheImunityGamer, 2024-05-10:
My main reason is organisation. I have a folder where I put my coding projects and it has spaces in it. I could make a separate folder like `Hardware_Design` but then the coding projects are getting more disorganized and that directory where they would be is already very disorganized. I guess I will need to do that if I want formal verification. I do wonder, doesn't SpinalSim do pretty much the same thing? I can generate random input (although its not quite as easy) and have a model for how the hardware should act, so shouldn't SpinalSim be enough?

As for a minimal example, I was testing with this:
```scala
object TopLevelFormal extends App {
  FormalConfig
    .withBMC(10)
    .doVerify(new Component {
      val dut = FormalDut(new Component() {
        val io = new Bundle {
          val state = out(UInt(8 bits))
          val cond0 = in(Bool())
        }

        val counter = Reg(UInt(8 bits)) init 0

        when(io.cond0) {
          counter := counter + 1
        }

        io.state := counter
      })

      // Ensure the formal test start with a reset
      assumeInitial(clockDomain.isResetActive)

      // Provide some stimulus
      formal.anyseq(dut.io.cond0)

      // Check the state initial value and increment
      assert(dut.io.state === past(dut.io.state + U(dut.io.cond0)).init(0))
    })
}
```

fayalalebrun, 2024-05-10:
In general, the main difference is that with formal verification you are proving your assertions to be true for all possible constrained inputs. If you are able to exhaustively try all inputs to your design, then that does indeed achieve the same thing. However, for designs that are not very simple this becomes unfeasible due to the huge number of possible combinations.

Trying a finite number of random inputs is not the same because you might miss some edge cases. However, it might be enough depending on your correctness requirements. For example, in my current project I am only formally verifying memory components and throughput/protocol rules on all of my` Stream` components.

PR #1412 (2024-05-10): Fix spaces in file paths for sby generation

Closes #1408 

# Context, Motivation & Description

The `sby` tool does not support spaces in paths. Since absolute paths are used in the "files" section of the .sby file, users whose projects are in paths with spaces cannot currently use SpinalHDL formal verification.

This fixes the issue by using relative paths instead of absolute paths. That is, as long as the component does not have a space in its name, but this shouldn't be support by VHDL/Verilog anyways.

# Impact on code generation

Files in the .sby "files" section use relative paths inste [... cut]
closed 2024-05-10 by Readon