SpinalHDL/SpinalHDL #1411: `privateNamespace` changes BlackBox component name

apstrike, 2024-05-09:
When using a BlackBox when `privateNamespace=true` in the `SpinalConfig`, a prefix is added to the blackboxed module on instantiation.

For example the following code:
```
class ExBB extends BlackBox {

  val io = new Bundle {
    val clk = in Bool()
    val dout = out Bool()
  }

  mapClockDomain(clock=io.clk)
}

class APExample extends Component {

  val io = new Bundle {
    val dout = out Bool()
  }

  val dut = new ExBB()

  io.dout := dut.io.dout

}

object TestConfig extends SpinalConfig(privateNamespace=true)

object Test {
  def main(args: Array[String]) {

    TestConfig.generateVerilog(new APExample())
  }
}
```

Produces the following Verilog file:
```
// Generator : SpinalHDL v1.10.1    git head : 2527c7c6b0fb0f95e5e1a5722a0be732b364ce43
// Component : APExample
// Git hash  : c48315fc0fad08e65876f209f9cbd59332330ae9

`timescale 1ns/1ps

module APExample (
  output wire          io_dout,
  input  wire          clk
);

  wire                dut_io_dout;

  APExample_ExBB dut (
    .io_clk  (clk        ), //i
    .io_dout (dut_io_dout)  //o
  );
  assign io_dout = dut_io_dout;

endmodule
```

I’d have assumed that blackboxed modules would keep their names as-is so that a user could share the module across module SpinalHDL projects.

andreasWallner, 2024-05-10:
If you use `privateNamespace` then all components are prefixed to prevent any possible clashes when integrating into another codebase.
If you use `globalPrefix` instead then Blackboxes won't be prefixed:
```scala
object TestConfig extends SpinalConfig(globalPrefix="APExample")
```
Does that work for you?

andreasWallner, 2024-06-14:
I guess that helped - closing after > 1 month

closed 2024-06-14 by andreasWallner

apstrike, 2024-06-17:
oh, I'm really sorry I honestly just completely forgot to respond.

`globalPrefix` does work in that it doesn't prefix the black boxes, but it does something I find off in which the global prefix is added to both the file name and the toplevel module.

so for example with:
```scala
object TestConfig extends SpinalConfig(globalPrefix="APExample")
```

The file created is `APExampleAPExample.v` and the toplevel module is `APExampleAPExample`. It is fine, but maybe not what was intended with `globalPrefix`?

Dolu1990, 2024-06-21:
> The file created is APExampleAPExample.v and the toplevel module is APExampleAPExample. It is fine, but maybe not what was intended with globalPrefix?

It is intended, the global prefix realy prefix everything to sandbox stuff completly.
for instance you can do :

```scala
xxxx.generateVerilog(new APExample.setDefinitionName("miaou))  
```

apstrike, 2024-06-21:
Good to know. Thank you

Strike Technologies, LLC (“Strike”) is part of the GTS family of companies. Strike is a technology solutions provider, and is not a broker or dealer and does not transact any securities related business directly whatsoever. This communication is the property of Strike and its affiliates, and does not constitute an offer to sell or the solicitation of an offer to buy any security in any jurisdiction. It is intended only for the person to whom it is addressed and may contain information that is privileged, confidential, or otherwise protected from disclosure. Distribution or copying of this communication, or the information contained herein, by anyone other than the intended recipient is prohibited. If you have received this communication in error, please immediately notify Strike at ***@***.***, and delete and destroy any copies hereof.
CONFIDENTIALITY / PRIVILEGE NOTICE: This transmission and any attachments are intended solely for the addressee. This transmission is covered by the Electronic Communications Privacy Act, 18 U.S.C ''2510-2521. The information contained in this transmission is confidential in nature and protected from further use or disclosure under U.S. Pub. L. 106-102, 113 U.S. Stat. 1338 (1999), and may be subject to attorney-client or other legal privilege. Your use or disclosure of this information for any purpose other than that intended by its transmittal is strictly prohibited, and may subject you to fines and/or penalties under federal and state law. If you are not the intended recipient of this transmission, please DESTROY ALL COPIES RECEIVED and confirm destruction to the sender via return transmittal.