SpinalHDL/SpinalHDL #1420: Assignment error with scala int `0x...` literal

NikLeberg, 2024-05-22:
Hi, am I doing something stupid?
If I do the following, then I get an `ASSIGNMENT ERROR`. But the literal `0x80000000` should perfectly fit into 32 bits, shouldn't it?

```scala
import spinal.core._
import spinal.core.sim._

case class Foo() extends Component {
  val io = new Bundle {
    val a = in port Bits(32 bits)
  }
}

object Bug extends App {
  SimConfig.compile(Foo()).doSim { dut =>
    dut.io.a #= 0x40000000 // ok
    dut.io.a #= 0x80000000 // ASSIGNMENT ERROR : -2147483648 is outside the range of io_a : Bits[32 bits]
  }
}
```

<details>
<summary>Full Backtrace</summary>

```
sbt:curlyrv> runMain curlyrv.Bug
[info] running (fork) curlyrv.Bug 
[info] [Runtime] SpinalHDL v1.10.1    git head : 2527c7c6b0fb0f95e5e1a5722a0be732b364ce43
[info] [Runtime] JVM max memory : 1984.0MiB
[info] [Runtime] Current date : 2024.05.22 12:28:17
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 0.354 : Checks and transforms
[info] [Progress] at 0.471 : Generate Verilog
[info] [Done] at 0.560
[info] [Progress] Simulation workspace in /workspaces/curly-engine/./simWorkspace/Foo
[info] [Progress] Verilator compilation started
[info] [Progress] Verilator compilation done in 2156.465 ms
[info] [Progress] Start Foo test simulation with seed 1783612449
[error] ASSIGNMENT ERROR : -2147483648 is outside the range of io_a : Bits[32 bits]
[info] [Error] Simulation failed at time=0
[error] Exception in thread "main" java.lang.Exception
[error]         at spinal.sim.SimError$.apply(Misc.scala:12)
[error]         at spinal.sim.DataType.rangeError(Signal.scala:13)
[error]         at spinal.sim.BitsDataType.checkLongRange(Signal.scala:78)
[error]         at spinal.sim.SimManager.setLong(SimManager.scala:178)
[error]         at spinal.core.sim.package$.setLong(package.scala:147)
[error]         at spinal.core.sim.package$SimBitVectorPimper.$hash$eq(package.scala:404)
[error]         at curlyrv.Bug$.$anonfun$new$2(Test.scala:15)
[error]         at curlyrv.Bug$.$anonfun$new$2$adapted(Test.scala:13)
[error]         at spinal.core.sim.SimCompiled.$anonfun$doSimApi$2(SimBootstraps.scala:610)
[error]         at spinal.sim.SimManager.threadBody$1(SimManager.scala:230)
[error]         at spinal.sim.SimManager.$anonfun$run$1(SimManager.scala:233)
[error]         at spinal.sim.SimThread.$anonfun$jvmThread$1(SimThread.scala:93)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 5 s, completed May 22, 2024, 12:28:19 PM
```
</details>

NikLeberg, 2024-05-22:
Oh, just noticed that it does not happen with `SInt`.. soo is scala implicitly converting it to a negative number first, and then tries to mush it into the `Bits` ? So then I see how this cannot work.
If I try it with the literal of `B"x40000000"` it dissalows it and tells me `overloaded method value #= [...] cannot be applied to (spinal.core.Bits)`

How am I supposed to set `a` with the uppermost bit set? Sorry I'm lost. 😅

Readon, 2024-05-22:
use BigInt？

Dolu1990, 2024-05-22:
scala / java are signed 32 bits integer by default.

NikLeberg, 2024-05-22:
That was it `0x80...L` did the trick, thanks

closed 2024-05-22 by NikLeberg