SpinalHDL/SpinalHDL #1426: Formal: support for temporal sequences

NikLeberg, 2024-05-24:
Hi there, not an issue more of a question.

Currently for formal verification SymbiYosys and Verilog/SystemVerilog are used. As documented [here](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Formal%20verification/index.html#formal-primitives) this is somewhat _limited_ to boolean `assert` `cover` and `assume` and some variations of `past` as this is what the underlying Yosys supports.

One really great thing for formal verification would be temporal sequences. Something like (stolen from [tomverbeure](https://tomverbeure.github.io/rtl/2019/01/04/Under-the-Hood-of-Formal-Verification.html#systemverilog-assertions)):
```systemverilog
sequence a_seq_b12_seq_c
    a ##1 b[*1:2] ## c
endsequence
```
This sequence results in a match when the following happens:
 - ‘a’ evaluates to 1
 - after that ‘b’ evaluates to 1 for one or for two consecutive clock cycles
 - after that, ‘c’ evaluates to 1

One can then as example do `assert(a_seq_b12_seq_c)` or `cover(a_seq_b12_seq_c)`.

Now, similar functionality is supported in the PSL language together with VHDL. GHDL supports a large range of PSL. GHDL can then be run as [Yosys plugin](https://github.com/ghdl/ghdl-yosys-plugin). And ontop formal verification with SymbiYosys is possible see: [`tmeissner/psl_with_ghdl`](https://github.com/tmeissner/psl_with_ghdl/tree/master).

For me this sounds like a possible solution to bring temporal sequences to SpinalHDL. Has somebody tested something in this regards? Maybe even worked on anything? I'd love to help in any way. But alone I don't think I could come up with a good solution.

Happy for any comments.
Kind regards, Nik

Readon, 2024-05-24:
The existing Formal facilities is based on VerilogHDL with SymbiYosys only.
There might be bunch of things to be added and tested.

NikLeberg, 2024-05-26:
> There might be bunch of things

What do you mean by that?
 - Are there existing things that are actively being worked on? Where could I tap in to support it?
 - Or is there nothing and SpinalHDL is simply open for contributions in this regards?

Readon, 2024-05-26:
> > There might be bunch of things
> 
> What do you mean by that?
> 
>     * Are there existing things that are actively being worked on? Where could I tap in to support it?

no, as I know.

you're welcome to commit PR.

> 
>     * Or is there nothing and SpinalHDL is simply open for contributions in this regards?

there are things to be consider first.
1. spinalhdl syntax for this
2. support for PSL generation
3. how to implement temperal sequence by verilog simulation.

tomverbeure, 2024-05-26:
Nik,

are you suggesting to create your own Scala based temporal assertion system? The number of primitives is not that large. You would skip a step and go immediately to converting that data structure from a non-deterministic FSM to a deterministic one, and then write out the Verilog for it. 

It wouldn't be syntax compatible with SV assertions or PSL, but the core methodology would be the same and the amount of work a lot less.

Tom

NikLeberg, 2024-05-26:
> there are things to be consider first.
> 
> 1. spinalhdl syntax for this
> 2. support for PSL generation
> 3. how to implement temporal sequence by verilog simulation.

Yeah, these things should be considered first. I'm not capable in doing this myself though.
As you/SpinalHDL seems to be very open to PRs I'll work on first bringing GHDL to the Yosys formal suite in SpinalHDL (I don't think this will be really difficult). How temporal sequences might be supported by SpinalHDL is at a later stage I'd propose.

NikLeberg, 2024-05-26:
Hi Tom! Avid reader of your blog!
I somehow found your post _Under the hood of Formal Verification_ linked above that lead me down a rabbit hole. As I was under the strong impression, that Yosys+GHDL supports temporal sequences already, I was quite stumped that plain Yosys (except the commercial one) does not.

As GHDL already does the heavy lifting for us (the conversion of non-deterministic FSM to a deterministic one etc.) I simply though about setting it up such that SpinalHDL can tap into that.

So what I'm suggesting is:
 - to implement/enable GHDL+Yosys for Formal workflows
 - to give SpinalHDL a way of expressing/building temporal sequences in scala language
 - generate equivalent PSL code of it
 - formally verify the PSL statements in SymbiYosys+GHDL

This would prevent us from reinventing the wheel of FSM conversion in scala. It's already done by GHDL.

Readon, 2024-05-26:
> Hi Tom! Avid reader of your blog! I somehow found your post _Under the hood of Formal Verification_ linked above that lead me down a rabbit hole. As I was under the strong impression, that [...]

So the thing might be generating PSL statement through SpinalHDL's generator, and write a backend for GHDL as SpinalHDL's backend, though.

BTW, Yosys's commercial version verific could support temporal sequences.

tomverbeure, 2024-05-26:
> * to give SpinalHDL a way of expressing/building temporal sequences in scala language
> * generate equivalent PSL code of it
> * formally verify the PSL statements in SymbiYosys+GHDL
> 
> This would prevent us from reinventing the wheel of FSM conversion in scala. It's already done by GHDL.

This sounds like a useful project, but one that requires good knowledge of temporal expressions and Scala, and a willingness to spend a lot of time writing software. I have none of those three! :-)

Tom

Readon, 2024-05-27:
There is a related discussion on this. #689

andreasWallner, 2024-05-27:
One quick note: I'm not sure that we should go down the route to basically drop verilog support (which we'd kind of do if we have temporal sequences in verification components in the standard lib, w/o support for them in verilog output). Adding GHDL support sounds great, adding sequences as well - but there should be a path to get them into SymbiYosys via verilog as well (maybe also generate SVA in the future).
Main reason for that thought it that I see for more products being used that use verilog/SVA then PSL (and therefore more know-how, tutorials, etc.)

Readon, 2024-05-28:
> One quick note: I'm not sure that we should go down the route to basically drop verilog support (which we'd kind of do if we have temporal sequences in verification components in the standard lib, [...]

I think a new property based way to describe PSL and SVA could be invovled. so that the existing code remains untouched. It could be run on parallel.

NikLeberg, 2024-05-28:
> It wouldn't be syntax compatible with SV assertions or PSL, but the core methodology would be the same and the amount of work a lot less.

@tomverbeure To understand your proposal, are you suggesting that it would be less time consuming to:
 - come up with a non standard way to define temporal sequences (one that is related to SVA and PSL but not really)
 - implementing the conversion of non-deterministic FSM in Spinal/Scala
 - writing the FSM as Verilog/VHDL gates
 - formally verify properties on that generated FSM with SymbiYosys

allexoll, 2024-05-28:
ah well that's an interesting discussion, I was thinking about it this this afternoon, and while someone better versed in formal verification might proove this wrong, I had a little play with my current understanding that sequences could be expressed with boolean logic. 
I looked at https://research.ibm.com/haifa/conferences/hvc2013/present/SvaFvTutorialHVC2013.pdf page 38 for some explaination of the sequence concat, and tried to make something similar:

```scala
object MyTopLevelFormal extends App {
  FormalConfig.withConfig(Config.formal).withDebug
    .withBMC(10)
    .withCover(10)
    .doVerify(new Component {
      val dut = FormalDut(new Component {
        val d = in Bool()
        val q = out Bool()
        q := Delay(d, 2)
      })
      // Ensure the formal test start with a reset
      assumeInitial(clockDomain.isResetActive)
      // Provide some stimulus
      anyseq(dut.d)
      // test assertion that a 1 on d should result in a 1 on q after 2 cycles
      assert(seq_concat(dut.d & pastValid(), 2, dut.q))
    })
  def pastValidAfterResetFor(cycles: Int): Bool = {
    // signalCache(ClockDomain.current -> "formal.pastValid")(RegNext(True) initial(False)  setWeakName("formal_with_past"))
    val past_valid_after_reset_for = signalCache(ClockDomain.current -> "formal.pastValidAfterReset")(past(!ClockDomain.current.isResetActive, cycles) init(False) setWeakName("formal_with_past_after_reset_for_".concat(cycles.toString)))
    past_valid_after_reset_for
  }

  def seq_concat(r: Bool, distance: Int, s: Bool)(implicit loc: Location): Bool = {
    require(distance >= 0, "Negative distance is not allowed in seq_concat")
    // r ##distance s
    // meaning, every t where s at t-d is true and s at t is true -> true
    val r_cond_d_ago = past(r, distance) & pastValidAfterResetFor(distance)
    // table:
    //    r: 0 1 0 1
    //    s: 0 0 1 1
    //    q: 1 0 1 1
    !r_cond_d_ago | s
  }
}
```

I don't know if this is the same as  assert(dut.d ##2 dut.r) but I think it is?

Largely, I'm unsure what would be the best way to deal with a more general case, especially the pastValid & reset reasoning, but I thought it might be somewhere to start. 

Obviously, correct my assumption if they are incorrect, I haven't been able to use temporal sequence for a while due to most tool that support it needing a license

allexoll, 2024-05-28:
nevermind, I read the article by tom and that's precisely it. there is still limitation, but I think it might offer some begining of composability with sequences as well. I guess it might be possible to constrain the size of the delay lines to the depth of the proof?

andreasWallner, 2024-06-01:
> I think a new property based way to describe PSL and SVA could be invovled. so that the existing code remains untouched. It could be run on parallel.

This sounds to me more like a research project (basically an alternative that encompasses both PSL and SVA).
There are two points here:
- Adding GHDL support: this is great, since it gives us the ability to have static verification for VHDL
- (especially for `lib`) requiring PSL seems like a mistake because it make it impossible to formally verify the verilog code that we are producing. For various reasons there are many places that use verilog and where verification needs to be done on the final HDL. If we require PSL this means they either have to switch to verilog, or can't use the formal verification since it would be for the generated VHDL, not the verilog.

Readon, 2024-06-02:
> ah well that's an interesting discussion, I was thinking about it this this afternoon, and while someone better versed in formal verification might proove this wrong, I had a little play with my [...]

Yes, some temporal sequence can be emulated. However, there are always and never keywords also, which is not easy to be emulated.

Readon, 2024-06-02:
> This sounds to me more like a research project (basically an alternative that encompasses both PSL and SVA). There are two points here:
>
> * Adding GHDL support: this is great, since it gives us the [...]

May be first support GHDL formal verification implementation first. Then add some extended syntax, so that it could be accepted by GHDL. and while generating to Verilog the compiler give out error. If later there is a solution of Verilog backend then add it up later.