SpinalHDL/SpinalHDL #1430: Formal proofs started to fail with newer versions of sby

fayalalebrun, 2024-06-01:
After updating to a newer version of yosys (Latest oss-cad-suite version), many of my proofs started to fail with the following error in the logfile:
`ERROR: $check cell $cover$anon.sv:42$9 with TRG_WIDTH > 1 is not support by async2sync, use clk2fflogic.`

Here is a small example which reproduces the issue:
```
object Formal extends App {
  FormalConfig.withDebug
    .withBMC(40)
    .withProve(40)
    .withCover(40)
    .doVerify(
      new Component {
        val a = Stream(UInt(1 bit))
        val r = Stream(UInt(1 bit))

        r << a
        cover(
          a.fire && r.fire && past(
            a.fire && r.fire && past(a.fire && r.fire && pastValid())
          )
        )
      }
    )
}

```

NikLeberg, 2024-06-01:
Try to add the config flag `withAsync` i.e. `FormalConfig.withDebug.withAsync` in there. I'm not exactly sure what it changes in Yosys/SymbiYosys (except that is turns multiclock mode on) but I got the same error message and got rid of it with that flag.

Readon, 2024-06-02:
> After updating to a newer version of yosys (Latest oss-cad-suite version), many of my proofs started to fail with the following error in the logfile: `ERROR: $check cell $cover$anon.sv:42$9 with [...]

How about add slave master description to a and r?

Readon, 2024-06-02:
it seems that you are verifing a totally combonatinal logic. so that it is recognized as async one by yosys.

fayalalebrun, 2024-06-02:
> Try to add the config flag `withAsync` i.e. `FormalConfig.withDebug.withAsync` in there. I'm not exactly sure what it changes in Yosys/SymbiYosys (except that is turns multiclock mode on) but I got the same error message and got rid of it with that flag.

This does not work, now I get an additional error:
```
[info] Traceback (most recent call last):
[info]   File "/usr/bin/sby", line 572, in <module>
[info]     taskloop.run()
[info]   File "/usr/bin/../share/yosys/python3/sby_core.py", line 591, in run
[info]     proc.poll()
[info]   File "/usr/bin/../share/yosys/python3/sby_core.py", line 170, in poll
[info]     self.poll(True)
[info]   File "/usr/bin/../share/yosys/python3/sby_core.py", line 211, in poll
[info]     self.read_output()
[info]   File "/usr/bin/../share/yosys/python3/sby_core.py", line 267, in read_output
[info]     self.handle_output(outs)
[info]   File "/usr/bin/../share/yosys/python3/sby_core.py", line 131, in handle_output
[info]     line = self.output_callback(line)
[info]   File "/usr/bin/../share/yosys/python3/sby_engine_smtbmc.py", line 243, in output_callback
[info]     prop = task.design.hierarchy.find_property_by_cellname(cell_name, trans_dict=smt2_trans)
[info]   File "/usr/bin/../share/yosys/python3/sby_design.py", line 146, in find_property_by_cellname
[info]     raise KeyError(f"No such property: {cell_name}")
[info] KeyError: 'No such property: _witness_.cover_cover_anon_sv_42_9'
```

> How about add slave master description to a and r?

Same error.

> it seems that you are verifing a totally combonatinal logic. so that it is recognized as async one by yosys. use
> 
> ```scala
>    r <-< a
> ```
> 
> the dut component would become sequential logic, then the yosys might recognize it.

Same error.

Readon, 2024-06-02:
Is there any reason that drive you use the latest version of sby? I am really hating to trace down the upstream's changes often.
How about using the fixed and tested version? 
There are several tools in this chain, sby, yosys, z4, yices2 and so on.

fayalalebrun, 2024-06-02:
I can use a previous version, but this might need to be fixed anyways in the future. It's also good to have this issue here so that other people who encounter it know to downgrade. I made an issue in the sby repo, let's see what they have to say.

Readon, 2024-06-02:
> I can use a previous version, but this might need to be fixed anyways in the future. It's also good to have this issue here so that other people who encounter it know to downgrade. I made an issue in the sby repo, let's see what they have to say.

Yes, on my machine yosys up to 0.33 is still working. 
I looked over the changelog of yosys, it seems that there is a change on 0.37 to 0.38 with chformal, async2sync and clk2fflogic for the function for $check which is not support by SpinalHDL at all, at least now.

Readon, 2024-06-03:
ok, the problem is that the new yosys synthizer might transform the asynchronous reset logic with cover or other code into a $check statement in its RTLIL, which is different from the previous one.

As discussed [here](https://github.com/YosysHQ/sby/issues/255#issuecomment-2032106247), we need to trun the multiclock on even when the reset is asynchronously assert. 
The only cases is that all reset and logic is used synchronously then multiclock could be off.

So the simple solution is to turn multiclock on by default.
More complex one is to detect all the logic of design, too see if there is asynchronous signal. if there is none then multiclock off.
It has to be done by checking all the clockdomian in the dut design.

Readon, 2024-06-03:
When multiclock is on the global clock is involved, so that the clocks have to be aligned to global clock by additional code, which makes the code bloat a lot.

I think it is not a good solution through.
It would be much easier to just verify a pure synchronous design by default.

Readon, 2024-06-03:
Both YosysHQ/yosys#4231 and amaranth-lang/amaranth#603 mentioned this.

Readon, 2024-06-04:
The verilog generated for formal verification should be in a fully synchronized logic or always check. like this.
``` Verilog
always @(posedge clk) begin
   ....
end
//or
always @(*) begin
   ...
end
```

The implementation now follows this mainly.
However, when we use rose(), fell() and such function. Now we emulate the behavior of those code by calling rise() and so on.
This code might lead to a normal logic code generating. 
So that an asynchronous code is generated, which will lead to bugs. 
In the old version of yosys, it ignored this and act almost correct. But in the latest version it seems that they fixed this.

One more thing is that the combinational logic pattern is required occasionally, if we could add this possibility to user? @Dolu1990 

may be with syntax like this
``` Scala
assume(xxx).alwaysOn
```

PR #1435 (2024-06-06): Verify component without withAsync option with synchronous reset as default.

Closes #1430

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?
commit 38637844 (2024-06-06): fix #1430.
closed 2024-06-06 by Readon