SpinalHDL/SpinalHDL #1431: Could you please advise if there is any method to generate a port in the format of [lsb:msb]?

hhucchenyixiao, 2024-06-02:
such as transforming `val a = Bits(8 bits)` into `wire [0:7] a;`

Dolu1990, 2024-06-04:
So, it would kinda only matter in places where you bring literals ? That impact the way binary literals are pushed to the bit vector ?

hhucchenyixiao, 2024-06-06:
> So, it would kinda only matter in places where you bring literals ? That impact the way binary literals are pushed to the bit vector ?

Yes, it would only affect the positioning of `a[index]` within vector `a`. Without this feature, an extra bit-reversal of the inputs would be needed before processing, and another one on the outputs afterwards.

Thank you very much for your clarification and patience in understanding this requirement.

Readon, 2024-06-07:
> Yes, it would only affect the positioning of `a[index]` within vector `a`. Without this feature, an extra bit-reversal of the inputs would be needed before processing, and another one on the outputs [...]

I think it would be better to keep the common sense bits order by default is good practice.
Explicitly reverse order as you wish seems better.

Only case is that if you want to write a module with IO in specific bits order, that would be fancy if there is a function could help.

Dolu1990, 2024-06-07:
I think i agree with keeping all the bit vector with the same sense.

Maybe the best for would be to create a simple class to represent fields literals with low:high string literal. And then having that class having a function to create hardware literals (high:low) ?

Else there is also https://github.com/SpinalHDL/SpinalCrypto/tree/master/crypto/src/main/scala/spinal/crypto