SpinalHDL/SpinalHDL #1433: Porting to spinalHDL

ztachip, 2024-06-02:
I would like to port my project in VHDL to spinalHDL.
Do we have a mapping between VHDL constructs (like records) to spinalHDL's constructs.
Is it an fairly automatic process or some redesign required.
Thanks

Dolu1990, 2024-06-03:
Hi,

Records can easily be remapped to Bundle class.
Here is some documentation related to VHDL <-> SpinalHDL : 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Getting%20Started/Help%20for%20VHDL%20people/vhdl_perspective.html

ztachip, 2024-06-04:
Great. It looks like porting  VHDL to Spinal is not very difficult.
However, some of my code, in the same VHDL file , uses 2 clock domain. How can I port this when clock is defined as domain in Spinal.
Thanks

https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Getting%20Started/Help%20for%20VHDL%20people/vhdl_perspective.html

Dolu1990, 2024-06-05:
Clockdomains are also supported in a quite nice way, check the doc ^^

closed 2024-06-05 by ztachip