SpinalHDL/SpinalHDL #1440: Porting SpinalHDL to CIRCT?

dobios, 2024-06-05:
Hi, 

[CIRCT](https://circt.llvm.org/) is a unified hardware compiler, part of the LLVM project, based on MLIR that supports many front-ends (including Magma and Chisel) and many back-ends (including SMTLib, btor2, and SystemVerilog), entirely in open-source.

I was wondering if there is any interest of porting SpinalHDL to CIRCT, as it would benefit both CIRCT and SpinalHDL. Unifying high-level hardware languages to use a single modularisable compiler would also benefit the advancement of the field in general.

Just a curiosity of mine, it's understandable if this isn't feasible given how developed SpinalHDL is, just wanted to know what people in this project think about this.

Thanks :)

Dolu1990, 2024-06-05:
Hi,

There has been some interrest, fondamentaly, adding another backend isn't a complicated/tricky task, but the main roadblock are : 
- The lack of manpower
- My potentiel lack of awareness of the advantages over the current VHDL / Verilog backends integrated in SpinalHDL, as i guess the SpinalHDL's verilog coult then be imported into CIRCT anyway ?
- SpinalHDL, unlike Chisel3 (as far as i know) keep the whole netlist AST in memory, and provide an API for the user to directly interact with it during elaboration time (read/modify). So a lot of netlist manipulation can be done easily without having to add another separated flow  (CIRCT), but isn't interroperable with non-spinalhdl things, which is one down side.  There is a quite incomplet doc about it here : https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Developers%20area/spinalhdl_datamodel.html

By the way, i will be at FSiC2024, in case you are around ^^
Cheers

dobios, 2024-06-05:
Hi thank you so much for your response !
The issues you bring up are what I was suspecting, but I'm glad to see that there is some interest in the SpinalHDL community.  

One benefit I can mention off the top of my head on the CIRCT front, is that it would allow for a very customizable verification experience and have you benefit from certain efforts that are happening on that end to support SVA/PSL constructs in open-source simulation and formal verification. It would also unlock some of the formal backends that afaik in SpinalHDL currently depends on external (well built but difficult to extend) tools such as Symbiyosys to get. 

These are just the first few things I can think of, I just think that a good approach for evolving this field is to have a unified central backend for certain general optimizations and targets, and then have front-ends be able to focus on specializing and offering a better experience for the users.

Either way, thank you for responding to my curiosity, I'm a big fan of the work SpinalHDL does and hope to explore what it offers more in the future. Thanks!

closed 2024-06-05 by dobios

Dolu1990, 2024-06-06:
> unified central backend for certain general optimizations and targets

I can imagine memory blackboxing / generation / integration in backend flow, those are a paine depending the targets ^^

> I'm a big fan of the work SpinalHDL does and hope to explore what it offers more in the future. Thanks!

Thanks :)
Cheers

Readon, 2024-06-07:
> Hi thank you so much for your response ! The issues you bring up are what I was suspecting, but I'm glad to see that there is some interest in the SpinalHDL community.
>
> One benefit I can mention off [...]

Hello, do you have a full open source tool chain that could run SVA verfication based on CIRCT?

dobios, 2024-06-07:
> Hello, do you have a full open source tool chain that could run SVA verfication based on CIRCT?
   
This is our current goal, we have some support for open-source simulation using SVA constructs with `arcilator` and some support for formal verification using `btor2` emission and `btormc`, or with `cirtct-mc` which uses Z3 as a backend. The way we currently support some SVA constructs is through property synthesis. This is still very much a work in progress, but will hopefully be rolled out soon.