SpinalHDL/SpinalHDL #1441: PackedBundle EndiannessSwap

oletf, 2024-06-06:
Hi, 

I'm currently trying to have a `PackedWordBundle` which contains an `UInt`, 
but I'd need this word's bytes endianness to be swapped in its packed form.

I've tried to have assignments done on a `signalCache` without much success...
```scala
case class TestPack() extends PackedWordBundle(32 bits) {
  private val word1BE = UInt(32 bits).packFrom(0).inWord(0)
  def word1 = signalCache(this, "word1")(EndiannessSwap(word1BE))
  //...other packed things
}
```
tried here to only use assignments on `word1` so the actual UInt word1BE gets the swapped signal, which is used once `packed` is called.
(getting `ASSIGNMENT OVERLAP` on the first assignment)

I probably missed an easier way to get what I need in Spinal, 
also not sure about how I should use signalCache in this situation if needed...

Any idea on how I could get this ?
 
thanks forward

andreasWallner, 2024-06-10:
As far as I can tell `EndianessSwap` is not from the SpinalHDL library so I have to guess a bit.

The most likely reason is that EndianessSwap returns a new signal that has the bits assigned with swapped endianess. So if you try to drive `myPack.word1 := ...` somewhere in your code you are not describing a driver for `word1BE` but a non-conditional driver for `word1` alone which conflicts with the driver inside `EndianessSwap`.

Many of the functions on UInt don't work "in both directions" (pretty much only the "Bit extraction" operations from the docs do). The `signalCache` does not help here as the issue already occurs on the first usage.

What's your use-case why you need the swapped endianess?
Do you just want to bit-order to change? `pack` has an optional parameter for this. If you want to byte-order to change this is currently not supported OOTB by PackedBundle as far as I know.

oletf, 2024-06-11:
> As far as I can tell `EndianessSwap` is not from the SpinalHDL library so I have to guess a bit.

it's from lib.Utils : [`EndianessSwap`](https://github.com/SpinalHDL/SpinalHDL/blob/987cc87b8eedf8a833373ca13b8f62ecc8ce1df6/lib/src/main/scala/spinal/lib/Utils.scala#L461-L477).

> Many of the functions on UInt don't work "in both directions" (pretty much only the "Bit extraction" operations from the docs do). The signalCache does not help here as the issue already occurs on the first usage.

Thanks for the insight, I'm still lacking understanding of internals of Spinal.

> What's your use-case why you need the swapped endianess?

I basically have a packed bundle which has a data in little-endian byte order, and I need both read and write, as it's used for both rx & tx. 
Indeed signalCache works well in the read only case.

I just wanted to avoid spraying `EndianessSwap` everywhere I read/write to this word before packing the bundle, but I guess I'll keep this for now !

Thanks, closing, because it was more to ask if I missed an obvious way to do this or not.

closed 2024-06-11 by oletf