SpinalHDL/SpinalHDL #1445: Support for AMBA ACE4

KireinaHoro, 2024-06-10:
I'm looking at adding some support for ACE 4 on top of the `amba4.axi` classes we already have in `spinal.lib`, but I'm not super sure on what's the best way to proceed here.  ACE adds two extra channels `AC` and `CR`, but also extra signals to `AR`, `AW` and `W` signals on the original AXI definition.  Should I

- derive `Axi4Ax` and `Axi4R` channels and add the ACE signals, or
- add flags like `useSnoop` on the `Axi4Ax` object?

Dolu1990, 2024-06-10:
I would say, adding withAce in the Axi4Config, then, hmm, are the signal added to AR and AW exactly the sames ?

KireinaHoro, 2024-06-10:
[image]

In a word, the snoop signals specify the read and write transactions as cache transactions that would also enforce the cache state of a memory location.  For example, a master can issue ReadShared in a read transaction via indicating it through `ARSNOOP`; the interconnect will broadcast ReadShared on the `AC` channel to masters in the same shareable domain to fulfill this requirement.

Dolu1990, 2024-06-10:
Ahhh sad XD
So, adding withAce to the Axi4Config, and adding all the related channels to Axi4, aswell as additional signals to Axi4Ar / Axi4Aw should be fine i think.  ?

Did you considered Tilelink (<3<3) ?

KireinaHoro, 2024-06-10:
Ah so basically not having a `amba4.ace`, but just use `Axi4` directly?

Yeah I wanted to, but I'm interfacing with the PS on a Zynq UltraScale+, which has an ACE port :/

Dolu1990, 2024-06-10:
> but just use Axi4 directly?

Yes
I mean in some way, you can see AXI ace as an extentions of AXI so, why not ^^

> which has an ACE port :/

Ahhhh damned XD
i didn't knew they were providing ace support to the fpga fabric XD
Do they specify which subset of AXI ACE they use / you should support ? 
Because that's a biiiiiiiiiiiiig spec with a loooooooooooooooot of variants.

KireinaHoro, 2024-06-11:
> i didn't knew they were providing ace support to the fpga fabric XD Do they specify which subset of AXI ACE they use / you should support ?

I think it's just the full ACE4 protocol, since the TRM of the Zynq US+ says that the other side of that port is the [ARM CCI-400](https://developer.arm.com/documentation/ddi0470/k/?lang=en), so it should support the full protocol.  The port could also be used as ACE-Lite, which disables the `AC`, `CR`, and `CD` channels, meaning that the PL master can snoop other masters (i.e. the APU in the Zynq), but cannot be snooped itself due to not having a cache.

I'm exploring the protocol at the moment, so yeah I embrace the complexities :) I'm trying to see how different it is from something like [ECI](https://arxiv.org/pdf/2208.07124) and if we can fit some stuff we did for ECI on top of ACE.

KireinaHoro, 2024-06-11:
I took a look at the channels and operations for TL-C: TileLink has a permission tree model, where as ACE has cache-line states and transitions.  However, the messages exchanged looks really similar (Probe vs CleanShared, etc.).

Dolu1990, 2024-06-11:
> so it should support the full protocol.

Be aware that generaly, AXI ACE isn't intended to be fully implemented.

> TileLink has a permission tree model,

To me, it is more a "syntax" difference than a real one.
Overal, AXI ace can implement transaction with more fine control compared with tilelink.
For instance, in tillink, when a master aquire a block, the L2 can't tell the master "Here is your block, and by the way, the data are dirty, so you need to give  it back to me when you want to get ride of it"
(ACE readClean vs readShared ..)

This kind of things

KireinaHoro, 2024-06-11:
I think the idea is that a snooped ACE master can handle messages differently, so long as it satisfies the required state change: 

[image]
[image]

So the simplest snooped master could handle all snoop transactions by writing back the line and sending `CRESP=00000`.  One could definitely build upon that...