SpinalHDL/SpinalHDL #1446: About convenient anonymous bundle.

Readon, 2024-06-10:
I am trying to practice on some full stream design, where need a bunch of interface data type between stream, almost each of the stream stage.
So that I want to define some internal data type in anonymous bundle as below, but not `case class` for all cases.
``` Scala
val chainIn = input.translateWith(new Bundle { val recipe = inRecipe; val totalLen = recipeTotalLen })
```
However, this code would lead to error "Spinal can't clone class XXXXX datatype".
According to some issues, it is caused by the clone is not clone able which is required by Stream.

Is there any other convenient way to define this?

Dolu1990, 2024-06-10:
> new Bundle { val recipe = inRecipe; val totalLen = recipeTotalLen }

This is tricky, Bundle are only made to define new types, with no backed-in connectivity.  There is currently no good syntax for all in one type declaration + initialization in SpinalHDL so far. You would need to define the type, then instanciate the type, then connect the type. That is quite a bit verbose.
Maybe scala macro could help ?

> where need a bunch of interface data type between stream, almost each of the stream stage.

Sound like a job for the lib.misc.pipeline API :) With it you will not have to define any "bundle" anywere.

Readon, 2024-06-11:
The problem is that how can I work with existing utilities like `StreamShiftChain`?

Readon, 2024-06-11:
Is it possible to get the input Stream and output Stream of a node, regardless on the cancel and forgot function^^

Dolu1990, 2024-06-11:
> The problem is that how can I work with existing utilities like StreamShiftChain?

That's where things get verbose , as you need to reconstruct things to/from bundles.
you  can use the Node arbitrateFrom arbitrateTo function to help with the arbitration.

Readon, 2024-06-11:
> > The problem is that how can I work with existing utilities like StreamShiftChain?
> 
> That's where things get verbose , as you need to reconstruct things to/from bundles. you can use the Node arbitrateFrom arbitrateTo function to help with the arbitration.

Emm, it seems that it is a little bit more sentences to be write.
And the type should be written explicitly. Such as the code in Doc, the payload of our up and down Streams is a little bit complex.
``` Scala
val up = slave Stream(UInt(16 bits))
val down = master Stream(UInt(16 bits)) //Note master Stream(OUT) is good aswell

n0.driveFrom(up)((self, payload) => self(IN) := payload)
n2.driveTo(down)((payload, self) => payload := self(OUT))
```

PR #1456 (2024-06-22): add driveTo method return stream directly.

Closes #1446
return stream directly by driveTo which simplify usage.

# Context, Motivation & Description

# Impact on code generation

# Checklist

- [ ] Unit tests were added
- [ ] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/SpinalDoc-RTD/issues/new?title=Document%20XXX&body=Do%20not%20merge%20until%20SpinalHDL/SpinalHDL%23XXX%20has%20not%20been%20merged)?