SpinalHDL/SpinalHDL #1451: switch generate redundant verilog code

cb38, 2024-06-12:
This can be seen generating .v  using the UartCtrlRxMain from the workshop. 

it generates multiples cases while one shall be enough. 
The code is valid though.

`
always @(*) begin
    bitCounter_clear = 1'b0;
    case(stateMachine_state)
      UartCtrlRxState_IDLE : begin
      end
      UartCtrlRxState_START : begin
        if(bitTimer_tick) begin
          bitCounter_clear = 1'b1;
        end
      end
      UartCtrlRxState_DATA : begin
      end
      default : begin
      end
    endcase
  end

  always @(*) begin
    io_read_valid = 1'b0;
    case(stateMachine_state)
      UartCtrlRxState_IDLE : begin
      end
      UartCtrlRxState_START : begin
      end
      UartCtrlRxState_DATA : begin
      end
      default : begin
        if(bitTimer_tick) begin
          io_read_valid = 1'b1;
        end
      end
    endcase
  end
`

Dolu1990, 2024-06-13:
Hi,

This is done on purpose.

The reason is that in some cases, verilog can create "simulation only" processing loop between unclocked always blocks if always blocks assign more than one signal in the unclocked always block.

There is an option in spinalhdl to merge always block with similar conditional scope.

cb38, 2024-06-13:
> There is an option in spinalhdl to merge always block with similar conditional scope.
what is the option ?

Dolu1990, 2024-06-14:
SpinalConfig(mergeAsyncProcess = true).generateVerilog(new MyToplevel)

But better not using it.

closed 2024-06-29 by cb38