SpinalHDL/SpinalHDL #1458: How to implement multiple stages in one Node by using the misc.pipeline utility.

Readon, 2024-06-17:
I have a Vec[T] type signal and want to call reduceBalancedTree on it, and insert several stages of pipeline.
When I put this logic into a pipeline Node, it seems that the Node is one stage concept, which could not provide multi-stage logic.
So the solution could be using driveTo to convert the Node output to a stream and reconstruct a reduceBalancedTree method for Stream[Vec[T]]. This solution might lose the synchronization of previous Nodes' payload.

I am curious about the best practice on the multiple stage implementation on pipeline Node concerned with the reduceBalancedTree.

saahm, 2024-06-19:
I'm not having a solution, but when I checked the implementation of reduceBalancedTree in Utils.scala there is a function called stage. 

```scala
def reduceBalancedTree(op: (T, T) => T, levelBridge: (T, Int) => T): T = {
  def stage(elements: ArrayBuffer[T], level: Int): T = {
    if (elements.length == 1) return elements.head
    val stageLogic = new ArrayBuffer[T]()
    val logicCount = (elements.length + 1) / 2

    for (i <- 0 until logicCount) {
      if (i * 2 + 1 < elements.length)
        stageLogic += levelBridge(op(elements(i * 2), elements(i * 2 + 1)), level)
      else
        stageLogic += levelBridge(elements(i * 2), level)
    }
    stage(stageLogic, level + 1)

  }
  val array = ArrayBuffer[T]() ++ pimped                                                                                                                              
  assert(array.nonEmpty, "can't use reduceBalancedTree on an empty collection")
  stage(array, 0)
}
```

I'm not entirely sure what levelBridge does, maybe it just takes care to wire up the stageLogic with the operator. It kinda reminds me of what happens in that one AdderTree example in the SpinalBootcamp ([Example: Parameter Pipline AdderTree](https://github.com/jijingg/Spinal-bootcamp/blob/dev/2.6-Spinal-core-Component-Function-Area.ipynb) in the lower half, but there it's using recursion).
Although these days we do have the pipeline API, so possibly there is a gap for a reduceBalancedTree function 'through' the pipeline API? It almost looks like that, having a pipelined version of "levelBridge" would be enough?

Readon, 2024-06-19:
I think levelBridge is not enough for this case. The Node natively is act in one stage, so the valid and ready would be managed only by one cycle case.
I have written one in #1459 , however with problem on elaborating.

Dolu1990, 2024-06-21:
I would say, reduceBalancedTree  fuse the notion of stage with the notion of data.
Meanwhile the misc.pipeline API seperate those two things very very explicitly.

So, what you can do, is to use the regular  reduceBalancedTree implementation, but instead of T being a Data or Payload, it should be a tupple to convey the notion of "what stage -> what data"

here is some pseudo code : 

```scala
val myNodes = Array.fill(10)(new Node)
val myInputs = List(x, y, z) // UInt
val myTupples = myInputs.map(v => 0 -> myNodes(0).insert(v)) // stageIndex -> Payload[UInt]
myInputs.spinalReduceBalanced(....)
```

Would that make sense ?

Then some more friendly api could be constructed on the top.

Readon, 2024-06-21:
I have tried one at #1459 . it seems that your code can simplify it.

closed 2024-10-05 by Readon