SpinalHDL/SpinalHDL #1462: Attributes on IO result in invalid syntax in VHDL

NikLeberg, 2024-06-21:
If the following code is ran:
```scala
import spinal.core._

case class Bug() extends Component {
  val io = new Bundle {
    val a = in port Bool()
  }
  io.a.addAttribute("foo")
}

object TopApp extends App {
  SpinalConfig.generateVhdl(Bug())
}
```

Then spinal generates invalid VHDL description:
```vhdl
entity Bug is
  port(
    io_a : in std_logic
  );
end Bug;

architecture arch of Bug is
  attribute foo : boolean;
  attribute foo of io_a : signal is true;
begin
end arch;
```

GHDL at least accepts the attribute if it is put in the `entity` section. I.e.:
```vhdl
entity Bug is
  port(
    io_a : in std_logic
  );
  attribute foo : boolean;
  attribute foo of io_a : signal is true;
end Bug;
```
Altought I dont know if this is valid VHDL or not. I could not find supporting information in the web.

Dolu1990, 2024-06-24:
I realy don't know if this is a GHDL missing feature, or if that is bad VHDL generated from SpinalHDL.
If somebody can find any additional evidence i can fix it.

Readon, 2024-06-24:
According to spec of VHDL, the entity and architecture can include attribute declaration and specification. However the attribute in example is for interface port, which might be put in entity.

Dolu1990, 2024-06-25:
fixed. Now : 

```scala
case class Bug() extends Component {
  val io = new Bundle {
    val a = in port Bool()
    val b = in port Bool()
  }
  io.a.addAttribute("foo")
  io.b.addAttribute("foo")
  io.b.addAttribute("asd")

  val x = Bool()
  val b = Bool()
  x.addAttribute("foo")
  b.addAttribute("asd")
}

```

=> 

```vhdl

entity Bug is
  port(
    io_a : in std_logic;
    io_b : in std_logic
  );
  attribute asd : boolean;
  attribute foo : boolean;

  attribute foo of io_a : signal is true;
  attribute foo of io_b : signal is true;
  attribute asd of io_b : signal is true;
end Bug;

architecture arch of Bug is
  attribute asd : boolean;
  attribute foo : boolean;

  signal x : std_logic;
  attribute foo of x : signal is true;
  signal b : std_logic;
  attribute asd of b : signal is true;
begin
end arch;
```

closed 2024-06-25 by Dolu1990

NikLeberg, 2024-06-26:
Thanks! Much appreciated.

commit 03906d9a (2024-06-25): Fix #1462
commit b6f6a35f (2024-06-25): Fix #1462